gin66/FastAccelStepper #98: Not working with esp-idf

rhr407, 2021-11-13:
I have done my best but have not been able to compile successfully as an idf component. I would highly appreciate if someone can have a look at the errors that I am facing below:

```
idf.py fullclean build
Executing action: fullclean
Executing action: all (aliases: build)
Running cmake in directory /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3"...
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/rizwan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/rizwan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/rizwan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/rizwan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/rizwan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/sdkconfig
-- Found PythonInterp: /home/rizwan/.espressif/python_env/idf4.4_py3.8_env/bin/python (found version "3.8.10") 
-- Found Perl: /usr/bin/perl (found version "5.30.0") 
-- App "Model-H" version: 1
-- Adding linker script /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: FastAccelStepper app_trace app_update arduino asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/components/FastAccelStepper /home/rizwan/esp/esp-idf/components/app_trace /home/rizwan/esp/esp-idf/components/app_update /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/components/arduino /home/rizwan/esp/esp-idf/components/asio /home/rizwan/esp/esp-idf/components/bootloader /home/rizwan/esp/esp-idf/components/bootloader_support /home/rizwan/esp/esp-idf/components/bt /home/rizwan/esp/esp-idf/components/cbor /home/rizwan/esp/esp-idf/components/cmock /home/rizwan/esp/esp-idf/components/coap /home/rizwan/esp/esp-idf/components/console /home/rizwan/esp/esp-idf/components/cxx /home/rizwan/esp/esp-idf/components/driver /home/rizwan/esp/esp-idf/components/efuse /home/rizwan/esp/esp-idf/components/esp-tls /home/rizwan/esp/esp-idf/components/esp32 /home/rizwan/esp/esp-idf/components/esp_adc_cal /home/rizwan/esp/esp-idf/components/esp_common /home/rizwan/esp/esp-idf/components/esp_eth /home/rizwan/esp/esp-idf/components/esp_event /home/rizwan/esp/esp-idf/components/esp_gdbstub /home/rizwan/esp/esp-idf/components/esp_hid /home/rizwan/esp/esp-idf/components/esp_http_client /home/rizwan/esp/esp-idf/components/esp_http_server /home/rizwan/esp/esp-idf/components/esp_https_ota /home/rizwan/esp/esp-idf/components/esp_https_server /home/rizwan/esp/esp-idf/components/esp_hw_support /home/rizwan/esp/esp-idf/components/esp_ipc /home/rizwan/esp/esp-idf/components/esp_lcd /home/rizwan/esp/esp-idf/components/esp_local_ctrl /home/rizwan/esp/esp-idf/components/esp_netif /home/rizwan/esp/esp-idf/components/esp_phy /home/rizwan/esp/esp-idf/components/esp_pm /home/rizwan/esp/esp-idf/components/esp_ringbuf /home/rizwan/esp/esp-idf/components/esp_rom /home/rizwan/esp/esp-idf/components/esp_serial_slave_link /home/rizwan/esp/esp-idf/components/esp_system /home/rizwan/esp/esp-idf/components/esp_timer /home/rizwan/esp/esp-idf/components/esp_websocket_client /home/rizwan/esp/esp-idf/components/esp_wifi /home/rizwan/esp/esp-idf/components/espcoredump /home/rizwan/esp/esp-idf/components/esptool_py /home/rizwan/esp/esp-idf/components/expat /home/rizwan/esp/esp-idf/components/fatfs /home/rizwan/esp/esp-idf/components/freemodbus /home/rizwan/esp/esp-idf/components/freertos /home/rizwan/esp/esp-idf/components/hal /home/rizwan/esp/esp-idf/components/heap /home/rizwan/esp/esp-idf/components/idf_test /home/rizwan/esp/esp-idf/components/ieee802154 /home/rizwan/esp/esp-idf/components/jsmn /home/rizwan/esp/esp-idf/components/json /home/rizwan/esp/esp-idf/components/libsodium /home/rizwan/esp/esp-idf/components/log /home/rizwan/esp/esp-idf/components/lwip /home/rizwan/esp/esp-idf/components/mbedtls /home/rizwan/esp/esp-idf/components/mdns /home/rizwan/esp/esp-idf/components/mqtt /home/rizwan/esp/esp-idf/components/newlib /home/rizwan/esp/esp-idf/components/nghttp /home/rizwan/esp/esp-idf/components/nvs_flash /home/rizwan/esp/esp-idf/components/openssl /home/rizwan/esp/esp-idf/components/openthread /home/rizwan/esp/esp-idf/components/partition_table /home/rizwan/esp/esp-idf/components/perfmon /home/rizwan/esp/esp-idf/components/protobuf-c /home/rizwan/esp/esp-idf/components/protocomm /home/rizwan/esp/esp-idf/components/pthread /home/rizwan/esp/esp-idf/components/sdmmc /home/rizwan/esp/esp-idf/components/soc /home/rizwan/esp/esp-idf/components/spi_flash /home/rizwan/esp/esp-idf/components/spiffs /home/rizwan/esp/esp-idf/components/tcp_transport /home/rizwan/esp/esp-idf/components/tcpip_adapter /home/rizwan/esp/esp-idf/components/tinyusb /home/rizwan/esp/esp-idf/components/ulp /home/rizwan/esp/esp-idf/components/unity /home/rizwan/esp/esp-idf/components/usb /home/rizwan/esp/esp-idf/components/vfs /home/rizwan/esp/esp-idf/components/wear_levelling /home/rizwan/esp/esp-idf/components/wifi_provisioning /home/rizwan/esp/esp-idf/components/wpa_supplicant /home/rizwan/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/build
Running ninja in directory /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/build
[... cut ...]
../components/FastAccelStepper/src/StepperISR_esp32.cpp: In member function 'int8_t StepperQueue::startPreparedQueue()':
../components/FastAccelStepper/src/StepperISR_esp32.cpp:418:23: error: 'volatile struct mcpwm_timer_regs_t' has no member named 'mode'
   mcpwm->timer[timer].mode.start = 2;  // 2=run continuous
                       ^~~~
[1417/1424] Building CXX object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/libraries/BLE/src/GeneralUtils.cpp.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

```

gin66, 2021-11-13:
Please try to use the branch [espidf4.4](https://github.com/gin66/FastAccelStepper/tree/espidf4.4).
This should be at least compatible to the new mcpwm definitions.

gin66, 2021-11-13:
The warning related to `fas_abs()` use is valid. Will fix it

gin66, 2021-11-13:
Fixed the warning and merged the branch espidf4.4  in master

rhr407, 2021-11-13:
Thank you so much for your help. I highly appreciate that. However, I am facing the following error now. My esp-idf is v4.4.

Generated /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/build/bootloader/bootloader.bin
[97/97] cd /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/build/bootloader/esp-idf/esptool_py && /home/rizwan/.espressif/python_env/idf4.4_py3.8_env/bin/python /home/rizwan/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v3/build/bootloader/bootloader.bin
Bootloader binary size 0x6350 bytes. 0xcb0 bytes (11%) free.
[1291/1425] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c-slave.c.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-i2c-slave.c.obj 
/home/rizwan/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino/variants/esp32 -I../components/arduino/cores/esp32 -I../components/arduino/libraries/ArduinoOTA/src -I../components/arduino/libraries/AsyncUDP/src -I../components/arduino/libraries/BLE/src -I../components/arduino/libraries/BluetoothSerial/src -I../components/arduino/libraries/DNSServer/src -I../components/arduino/libraries/EEPROM/src -I../components/arduino/libraries/ESP32/src -I../components/arduino/libraries/ESPmDNS/src -I../components/arduino/libraries/FFat/src -I../components/arduino/libraries/FS/src -I../components/arduino/libraries/HTTPClient/src -I../components/arduino/libraries/HTTPUpdate/src -I../components/arduino/libraries/LittleFS/src -I../components/arduino/libraries/NetBIOS/src -I../components/arduino/libraries/Preferences/src -I../components/arduino/libraries/RainMaker/src -I../components/arduino/libraries/SD_MMC/src -I../components/arduino/libraries/SD/src -I../components/arduino/libraries/SimpleBLE/src -I../components/arduino/libraries/SPIFFS/src -I../components/arduino/libraries/SPI/src -I../components/arduino/libraries/Ticker/src -I../components/arduino/libraries/Update/src -I../components/arduino/libraries/USB/src -I../components/arduino/libraries/WebServer/src -I../components/arduino/libraries/WiFiClientSecure/src -I../components/arduino/libraries/WiFi/src -I../components/arduino/libraries/WiFiProv/src -I../components/arduino/libraries/Wire/src -I../components/arduino/cores/esp32/libb64 -I/home/rizwan/esp/esp-idf/components/newlib/platform_include -I/home/rizwan/esp/esp-idf/components/freertos/include -I/home/rizwan/esp/esp-idf/components/freertos/include/esp_additions/freertos -I/home/rizwan/esp/esp-idf/components/freertos/port/xtensa/include -I/home/rizwan/esp/esp-idf/components/freertos/include/esp_additions -I/home/rizwan/esp/esp-idf/components/esp_hw_support/include -I/home/rizwan/esp/esp-idf/components/esp_hw_support/include/soc -I/home/rizwan/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/rizwan/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/rizwan/esp/esp-idf/components/heap/include -I/home/rizwan/esp/esp-idf/components/log/include -I/home/rizwan/esp/esp-idf/components/lwip/include/apps -I/home/rizwan/esp/esp-idf/components/lwip/include/apps/sntp -I/home/rizwan/esp/esp-idf/components/lwip/lwip/src/include -I/home/rizwan/esp/esp-idf/components/lwip/port/esp32/include -I/home/rizwan/esp/esp-idf/components/lwip/port/esp32/include/arch -I/home/rizwan/esp/esp-idf/components/soc/include -I/home/rizwan/esp/esp-idf/components/soc/esp32/. -I/home/rizwan/esp/esp-idf/components/soc/esp32/include -I/home/rizwan/esp/esp-idf/components/hal/esp32/include -I/home/rizwan/esp/esp-idf/components/hal/include -I/home/rizwan/esp/esp-idf/components/hal/platform_port/include -I/home/rizwan/esp/esp-idf/components/esp_rom/include -I/home/rizwan/esp/esp-idf/components/esp_rom/include/esp32 -I/home/rizwan/esp/esp-idf/components/esp_rom/esp32 -I/home/rizwan/esp/esp-idf/components/esp_common/include -I/home/rizwan/esp/esp-idf/components/esp_system/include -I/home/rizwan/esp/esp-idf/components/esp_system/port/soc -I/home/rizwan/esp/esp-idf/components/esp_system/port/public_compat -I/home/rizwan/esp/esp-idf/components/esp32/include -I/home/rizwan/esp/esp-idf/components/xtensa/include -I/home/rizwan/esp/esp-idf/components/xtensa/esp32/include -I/home/rizwan/esp/esp-idf/components/driver/include -I/home/rizwan/esp/esp-idf/components/driver/esp32/include -I/home/rizwan/esp/esp-idf/components/esp_pm/include -I/home/rizwan/esp/esp-idf/components/esp_ringbuf/include -I/home/rizwan/esp/esp-idf/components/efuse/include -I/home/rizwan/esp/esp-idf/components/efuse/esp32/include -I/home/rizwan/esp/esp-idf/components/vfs/include -I/home/rizwan/esp/esp-idf/components/esp_wifi/include -I/home/rizwan/esp/esp-idf/components/esp_event/include -I/home/rizwan/esp/esp-idf/components/esp_netif/include -I/home/rizwan/esp/esp-idf/components/esp_eth/include -I/home/rizwan/esp/esp-idf/components/tcpip_adapter/include -I/home/rizwan/esp/esp-idf/components/esp_phy/include -I/home/rizwan/esp/esp-idf/components/esp_phy/esp32/include -I/home/rizwan/esp/esp-idf/components/esp_ipc/include -I/home/rizwan/esp/esp-idf/components/app_trace/include -I/home/rizwan/esp/esp-idf/components/esp_timer/include -I/home/rizwan/esp/esp-idf/components [... cut ...]
../components/arduino/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2c_slave_isr_handler':
../components/arduino/cores/esp32/esp32-hal-i2c-slave.c:694:13: warning: unused variable 'tx_fifo_len' [-Wunused-variable]
     uint8_t tx_fifo_len = SOC_I2C_FIFO_LEN - i2c_ll_get_txfifo_len(i2c->dev);
             ^~~~~~~~~~~
[1296/1425] Building C object esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-sigmadelta.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

gin66, 2021-11-13:
I cannot see an error related to,FastAccelStepper. Looks like the esp-idf is buggy.

rhr407, 2021-11-13:
I think this problem pertains to the arduino component. which esp-idf and arduino component version are you using please?

gin66, 2021-11-13:
I do not use esp-idf. Normally I use only platformio.
In order to check for esp-idf4.4, I have fired up the Arduino-IDE with this [board definitions](https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json), installed version 2.0.1 and compile for "ESP32 Dev Module"

So it should be:
```
        {
          "name": "esp32",
          "architecture": "esp32",
          "version": "2.0.1",
          "category": "ESP32",
          "url": "https://github.com/espressif/arduino-esp32/releases/download/2.0.1/esp32-2.0.1.zip",
          "archiveFileName": "esp32-2.0.1.zip",
          "checksum": "SHA-256:3a7cd46ba47990dd37fbe02b7f0a910dd5cc7af1d190350b69d320ed36cd6b41",
          "size": "148976301",
          "help": {
            "online": ""
          },
          "boards": [
            {
              "name": "ESP32 Dev Board"
            },
            {
              "name": "ESP32-S2 Dev Board"
            },
            {
              "name": "ESP32-C3 Dev Board"
            }
          ],
          "toolsDependencies": [
            {
              "packager": "esp32",
              "name": "riscv32-esp-elf-gcc",
              "version": "gcc8_4_0-esp-2021r2"
            },
            {
              "packager": "esp32",
              "name": "xtensa-esp32-elf-gcc",
              "version": "gcc8_4_0-esp-2021r2"
            },
            {
              "packager": "esp32",
              "name": "xtensa-esp32s2-elf-gcc",
              "version": "gcc8_4_0-esp-2021r2"
            },
            {
              "packager": "esp32",
              "name": "esptool_py",
              "version": "3.1.0"
            },
            {
              "packager": "esp32",
              "name": "mkspiffs",
              "version": "0.2.3"
            },
            {
              "packager": "esp32",
              "name": "mklittlefs",
              "version": "3.0.0-gnu12-dc7f933"
            }
          ]
        },

```

rhr407, 2021-11-13:
Thanks a lot for your help. The issue is resolved. 

Solution: I have used idf-release/v4.4 branch for Arduino component and espidf4.4 from the  FastAccelStepper repo.

gin66, 2021-11-16:
@rhr407 Could you please check, if the branch espidf allows compilation without Arduino-component ?

gin66, 2021-11-16:
You may need to update CMakeLists.txt in the root directory to not require the Arduino library

rhr407, 2021-11-16:
Sure, I can check. Thanks for your help.

rhr407, 2021-11-16:
I think, it does not work without arduino library. Here is the output:

model-h-idf-fastaccelstepper$ idf.py flash monitor
Executing action: flash
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting.....
Detecting chip type... ESP32
Running ninja in directory /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- Component directory /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/RMaker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/TMCSTEPPER does not contain a CMakeLists.txt file. No component will be added
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/sdkconfig
-- App "Model-H" version: 1.2
-- Adding linker script /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /home/rizwan/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
ESP Insights Project commit: 
ESP RainMaker Project commit: 6b8c30c-dirty
-- Components: FastAccelStepper app_insights app_reset app_trace app_update app_wifi arduino asio bootloader bootloader_support bt button cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_diagnostics esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_insights esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_rainmaker esp_ringbuf esp_rom esp_schedule esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json json_generator json_parser libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread qrcode rmaker_common rtc_store sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant ws2812_led xtensa
-- Component paths: /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/FastAccelStepper /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/app_insights /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/app_reset /home/rizwan/esp/esp-idf/components/app_trace /home/rizwan/esp/esp-idf/components/app_update /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/app_wifi /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/arduino /home/rizwan/esp/esp-idf/components/asio /home/rizwan/esp/esp-idf/components/bootloader /home/rizwan/esp/esp-idf/components/bootloader_support /home/rizwan/esp/esp-idf/components/bt /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/button /home/rizwan/esp/esp-idf/components/cbor /home/rizwan/esp/esp-idf/components/cmock /home/rizwan/esp/esp-idf/components/coap /home/rizwan/esp/esp-idf/components/console /home/rizwan/esp/esp-idf/components/cxx /home/rizwan/esp/esp-idf/components/driver /home/rizwan/esp/esp-idf/components/efuse /home/rizwan/esp/esp-idf/components/esp-tls /home/rizwan/esp/esp-idf/components/esp32 /home/rizwan/esp/esp-idf/components/esp_adc_cal /home/rizwan/esp/esp-idf/components/esp_common /mnt/sda2/Insync/rhr407@gmail.com/Google-Drive/SEMPHOS/Projects/OnGoing/ValarSystems/Rizwan/Project_v5/model-h-idf-fastaccelstepper/components/esp_diagnostics /home/rizwan/esp/esp-idf/components/esp_eth /home/rizwan/esp/esp-idf/components/esp_event /home/rizwan/esp/esp-idf/components/esp_gdbstub /home/rizwan/esp/esp-idf/components/esp_hid /home/rizwan/esp/esp-idf/components/esp_http_client /home/rizwan/esp/esp-idf/componen [... cut ...]
../components/FastAccelStepper/src/StepperISR.h:379:5: note: suggested alternative: 'interrupts'
     fasDisableInterrupts();
     ^~~~~~~~~~~~~~~~~~~~
     interrupts
../components/FastAccelStepper/src/StepperISR.h:382:5: error: 'fasEnableInterrupts' was not declared in this scope
     fasEnableInterrupts();
     ^~~~~~~~~~~~~~~~~~~
../components/FastAccelStepper/src/StepperISR.h:382:5: note: suggested alternative: 'noInterrupts'
     fasEnableInterrupts();
     ^~~~~~~~~~~~~~~~~~~
     noInterrupts
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

gin66, 2021-11-17:
Thanks for testing. Similar errors I have already fixed using platformio with espidf, but compilation under idf seems to be still different. Seems I need to install the toolchain to debug this.

gin66, 2022-10-05:
Thanks for the info and I will later clone your repo with the test project. Hope I can make it run under github actions as testbed

JustinOng, 2022-10-06:
It would be great to maintain IDF compatbility!

Upon further testing, it seems like commenting out `SUPPORT_ESP32_MCPWM_PCNT` is not a good way to test the operation of the RMT. Initialising six stepper objects by calling `stepperConnectToPin` with dummy pins, then one more object connected to my actual stepper does turn the stepper, so RMT support is also functional.

gin66, 2023-05-13:
the version 0.29.2 allows to directly selected the rmt module.
0.29.2 will be released in a couple of days…giving arduino library manager time to clarify their issue with releases.

closed 2023-05-13 by gin66