gin66/FastAccelStepper #100: ESP32-S3 Series

gin66, 2021-11-14:
This device contains mcpwm module, but only 4 instead of 8 pulse counters.
Thus 4 steppers should be possible to be supported.

Anyone available for test, so an update of the driver makes sense ?

doudar, 2022-04-08:
Sure, I'm available to test if you like. 

Just tried a quick compile for S3 from master on an existing large project ([SmartSpin2k](https://github.com/doudar/SmartSpin2k)). Relevant platformio.ini:

`platform = https://github.com/tasmota/platform-espressif32/archive/refs/heads/IDF44/ESP32-S3.zip`
`board = esp32-s3-devkitc-1`
`framework = arduino`

errors (Probably because you haven't implemented anything yet):

`.pio/libdeps/release/FastAccelStepper/src/common.h:150:2: error: #error "Unsupported derivate"
 #error "Unsupported derivate"
  ^~~~~
In file included from .pio/libdeps/release/FastAccelStepper/src/common.h:162,
                 from .pio/libdeps/release/FastAccelStepper/src/FastAccelStepper.h:5,`

`.pio/libdeps/release/FastAccelStepper/src/common.h:168:21: error: 'QUEUES_MCPWM_PCNT' was not declared in this scope
 #define NUM_QUEUES (QUEUES_MCPWM_PCNT + QUEUES_RMT)`

`.pio/libdeps/release/FastAccelStepper/src/common.h:168:41: error: 'QUEUES_RMT' was not declared in this scope
 #define NUM_QUEUES (QUEUES_MCPWM_PCNT + QUEUES_RMT)`

gin66, 2022-04-11:
Thanks for your willingness to test on esp32-s3. Please be patient, because these days I am quite busy. Hope to start on the adaptation within the next two weeks.

As the driver already supports rmt in addition to mcpwm+pcnt. These means, two possible paths to follow. During the adaptation will see, which path generates the least effort.

martinmay67, 2022-11-11:
Hi guys, any progress on this? At least idea about which direction to go.

martinmay67, 2022-11-16:
Okay, I dis some changes and have a version that works with ESP32S3. Should I create PR?

gin66, 2022-11-16:
sure. Looking forward to this

HeldeReis, 2023-01-22:
> Okay, I dis some changes and have a version that works with ESP32S3. Should I create PR?

Hello Martin,

Can you share your modifications?
I will be migrating a project soon to the esp32-s3 and it would be of great help.

HeldeReis, 2023-03-02:
Hello gin66,

I made some changes to allow support to ESP32 S3 and made a pullrequest (ESP32 S3 MCPWM/PCNT Support #162).
It is the first time that i make something on gihub so i hope not to have missed something.
I am available if you need some specific tests to be done.

gin66, 2023-03-02:
Thanks for the patch. I have merged in and the github action to build the sw has worked.
So I release it with 0.28.4

closed 2023-03-02 by gin66