gin66/FastAccelStepper #99: Stopping motion after runForward() or   runBackward()

cculbreath, 2021-11-14:
Hi there. Thanks for a great library. I have one quick question: is stopMove() the intended method for ending the continuous motion initiated by runForward() or runBackward()?

gin66, 2021-11-14:
`stopMove()` similarly like `runForward()` or `runBackward()` do not take care for the current or any intended target position of the stepper motor. Based on this it fits to ending the continuous motion.

Nevertheless it depends on the use case. Here are a few ideas for illustration:
- If the absolute stepper position is not known in the application, but there is a zero position sensor, then there is the possibility to initiate `runBackward()`with slow speed and if the zero position sensor is triggered, then to use `stopMove()` (stop with deceleration) or even `forceStopAndNewPosition(0)` (immediate stop).
- If the position sensor is placed at position 1000, then the continuous motion initiated by `runBackward()` could be stopped with a call to `move(-1000);setPositionAfterCommandsCompleted(0);`. The stepper will still run approximately (!) to zero position and the absolute stepper position is set to 0.
- Another use case: The stepper should run for a while at slow speed (100 steps/s) until e.g. the user presses a "zero" button and then the stepper should return with high speed (10000 steps/s)  to zero position. In this case the continuous movement could be stopped by just `setSpeedInHz(10000);moveTo(0);setSpeedInHz(100);`. Note: The last `setSpeedInHz()` will not be applied to the `moveTo()`!

In other words, the continuous motion is stopped with one of:
- `move()`
- `moveTo()`
- `stopMove()`
- `forceStopAndNewPosition()`

cculbreath, 2021-11-14:
Great. That all makes sense. Thank you for the quick reply!

closed 2021-11-14 by cculbreath