gin66/FastAccelStepper #101: Survey: esp32-s2 and esp32-c3 support

gin66, 2021-11-15:
In the espresso issue tracker, it was remarked, that the RMT module could be used for step generation, too
[reference](https://github.com/espressif/esp-idf/issues/7890). 

This would allow stepper control for the two derivates s2/c3 without mcpwm module.

Question is:  Anyone need support s2/c3 derivate and can help with test on those devices ?

LioRei, 2022-02-01:
Hi,

I actually have a system who operate a door in an escape game with this library. The whole setup is controlled by esp32s2 but with an arduino in the middle to be able to control the stepper (I2C communication between esp32s2 and arduino).

If you want, I can try the lib with my setup, it will simplify my work by removing the arduino ;)

gin66, 2022-02-01:
None of the platformio listed boards is supported by Arduino,
```
featheresp32-s2
esp32-s2-kaluga-1
esp32-s2-saloa-1
esp32-s2-franzininfo
```

And the last one - despite listed - cannot be found, which is funny.

Anyway, do you use espidf ?

LioRei, 2022-02-02:
Oh yes.... Since the 2.0.0 of esp32-arduino (which support esp32-s2 and c3), they (expressif) have broke the support of plateformio... So, I use esp32-arduino but with a plug-in of visual studio

LioRei, 2022-02-02:
For more information https://github.com/espressif/arduino-esp32/issues/6044

gin66, 2022-02-02:
Thanks for info.

In the meantime esp32 code has been refactored to separate the mcpwm/pcnt-functionality. This is preparation for the esp32s2 code. esp32s2 and esp32c3 platformio-targets have been added and compilation is successful.... just will do nothing.

So ready for the next step to add the rmt-stuff. Guess this will take some more time

gin66, 2022-02-04:
Initial support is available based on esp32. The esp32s2 version compiles at least.

The code is not much tested on esp32 and still may buggy. Nevertheless good enough to check, if it works on esp32s2.

LioRei, 2022-02-04:
I've received my esp32-c3 too so I will test this week end. I will keep you updated

gin66, 2022-02-04:
Just have checked the esp32-c3 rmt-spec. There are several changes on the module, so I doubt, that I can add support without having a real device. Besides - even on esp32 - the spec of the rmt module misses couple of details. So I had to change the approach for the implementation more than once. The current implementation looks ok and need now to get the remaining bugs identified and fixed.

So let's focus on S2, which seems to be compatible enough to esp32.

gin66, 2022-02-06:
So… apparently does not work yet. Need more time. Please wait

LioRei, 2022-02-06:
I haven't be able to test it yet so take your time, I'm not hurry ;)

gin66, 2022-02-06:
Just have pushed out version 0.27.0. On esp32 the rmt-backed stepper has passed the (limited) test suite, which ensures base functionality. Now I hope, that esp32s2 works, too.

LioRei, 2022-02-07:
[Mafia-Bibliotheque.txt](https://github.com/gin66/FastAccelStepper/files/8014881/Mafia-Bibliotheque.txt)
I've just tried the lib and.... it's working but with bug, I explain:

In my sketch, I move 10mm forward, backward until hitting my limit switch and forward again until not hitting limit switch anymore (a pretty standard homing ). The stepper go forward, go backward but is not moving anymore after this. As I can see, the enable pin is still set even when the stepper doesn't need to move anymore.

You will need to change the extension of the file to ino as github doesn't allowed me to upload an ino file

gin66, 2022-02-07:
Thanks for the report. Based on your description, I have quickly found out, why the motor cannot start again. There is a flag, which prevents to restart the queue, while the driver is somehow cleaning up after queue stop. Just the forceStop() has not reset this flag, so the queue could not be restarted and the motor will not move again.

What has puzzled me: The test on esp32 works, which is more than a surprise based on above finding. So I have done some more investigation: The root cause is, that the esp32s2 rmt module can be stopped, while esp32 only can run till it hits the end of the buffer.
The first was, what my code has intended. And the second was, what the code has done. Just hitting the buffer end, has triggered an interrupt, which then cleared the flag. So I could not find that in my code.

Just have pushed out 0.27.4, which should work as expected.

LioRei, 2022-02-07:
Ok, on my test bench the stepper is stepping well. I will test it on my production tomorrow ;)

LioRei, 2022-02-08:
I confirm, it's working well on my production too !

Thx a lot :)

gin66, 2022-02-08:
nice. Thanks for the confirmation

closed 2022-02-08 by gin66