gin66/FastAccelStepper #102: Not compiling with arduino-esp32.git#2.0.1

doudar, 2021-11-22:
Final linking fails when using framework arduino-esp32.git#2.0.1

Here are the warnings:

`.pio/libdeps/release/FastAccelStepper/src/StepperISR_esp32_espidfv4.4.cpp: In member function 'void StepperQueue::init(uint8_t, uint8_t)':
.pio/libdeps/release/FastAccelStepper/src/StepperISR_esp32_espidfv4.4.cpp:298:62: warning: 'esp_err_t mcpwm_isr_register(mcpwm_unit_t, void (*)(void*), void*, int, intr_handle_data_t**)' is deprecated: interrupt events are handled by driver, please use callback [-Wdeprecated-declarations]
         NULL, ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED, NULL);
                                                              ^
In file included from .pio/libdeps/release/FastAccelStepper/src/StepperISR.h:58,
                 from .pio/libdeps/release/FastAccelStepper/src/StepperISR_esp32_espidfv4.4.cpp:2:
C:/Users/Anthony/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/mcpwm.h:926:11: note: declared here
 esp_err_t mcpwm_isr_register(mcpwm_unit_t mcpwm_num, void (*fn)(void *), void *arg, int intr_alloc_flags,
           ^~~~~~~~~~~~~~~~~~
.pio/libdeps/release/FastAccelStepper/src/StepperISR_esp32_espidfv4.4.cpp:298:62: warning: 'esp_err_t mcpwm_isr_register(mcpwm_unit_t, void (*)(void*), void*, int, intr_handle_data_t**)' is deprecated: interrupt events are handled by driver, please use callback [-Wdeprecated-declarations]
         NULL, ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_SHARED, NULL);
                                                              ^
In file included from .pio/libdeps/release/FastAccelStepper/src/StepperISR.h:58,
                 from .pio/libdeps/release/FastAccelStepper/src/StepperISR_esp32_espidfv4.4.cpp:2:
C:/Users/Anthony/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/mcpwm.h:926:11: note: declared here
 esp_err_t mcpwm_isr_register(mcpwm_unit_t mcpwm_num, void (*fn)(void *), void *arg, int intr_alloc_flags,`

gin66, 2021-11-22:
The warnings should not be the cause for not linking. Is there any linking error ?

For the deprecation warning: This has been fixed by [e6ee8b2](https://github.com/espressif/esp-idf/commit/e6ee8b2fcf0ba3d9f7ec4654ceb0d61fb4857da1) in the esp-idf. See [#7890](https://github.com/espressif/esp-idf/issues/7890) for reference

doudar, 2021-11-22:
Yah, unfortunately it only says “final linking failed”. 

The large warning message just made me suspect this library, but it’s a large project so I’ll try commenting out this library (and associated code) and see if it compiles.

doudar, 2021-11-23:
It looks like the issue was that 2.0.1 is just too new with PlatformIO. following this: https://github.com/espressif/arduino-esp32/issues/5436

Then adding (modifying) the following to my platformio.ini fixed the issue:
`platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`
`platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.1
     toolchain-xtensa32@~2.80400.0`

Feel free to close the issue once you see it. 

Thanks!

closed 2021-11-24 by gin66