gin66/FastAccelStepper #105: I have problem stopping stepper motor

quangbinhhvac, 2022-01-08:
Hi gin66,
I have problem stopping stepper motor
"
void trucZlen() {
  if (stepperZ) {
     stepperZ->setSpeedInHz(1*800);       // steps/s
     stepperZ->setAcceleration(1*10*800);    // steps/s²
     stepperZ->runForward();
     if ((digitalRead(up) == 1)&&(stepperZ->isRunning() == true)) { //nếu thả nút và step đang chạy
     stepperZ->forceStopAndNewPosition(0);
     }
  }
}
"
Please help me
Thank you!

gin66, 2022-01-09:
Just for illustration, I have modified your code a bit
```
void trucZlen() {
  if (stepperZ) {
   // we have an initialized stepper, good
   // are we already at zero position ?
   if (digitalRead(up) != 1) {
     // no, so run towards it 
     stepperZ->setSpeedInHz(800); // steps/s
     stepperZ->setAcceleration(8000); // steps/s
     stepperZ->runForward(); 
     // BTW: runBackward() makes more sense,
     // unless you want to work with negative positions

     // stepper is started
     if ((digitalRead(up) != 1)&&(stepperZ->isRunning() == true)) {
         // stepper is running and we are still not at zero position
         // assume zero position is digitalRead(up) == 1
         // so wait and check again
         // BTW: the check for isRunning() is obsolete
     }
     // stepper just has reached zero position. So perform immediate stop
     stepperZ->forceStopAndNewPosition(0);
    }
  }
}
```

quangbinhhvac, 2022-01-10:
Hi gin66,
I have a problem when running at low speed with runforward and runback command, the motor is quite vibrating and noisy
I do not use “stepperZ->forceStopAndNewPosition(0);”  to stop the engine

gin66, 2022-01-10:
Do you use micro-stepping ?

quangbinhhvac, 2022-01-10:
yes i use 1/4

gin66, 2022-01-10:
Then please share the code of the project

quangbinhhvac, 2022-01-11:
#include <Wire.h> //khai báo thư viện
#include <EEPROM.h> //khai báo thư viện
#include "FastAccelStepper.h"
#include "AVRStepperPins.h" // Only required for AVR controllers
FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepperZ = NULL;
#define up A0
#define down 12
#define enablePin1 A4
#define dirPin1 8
#define stepPin1 9
//...............................................
void setup() {
  Serial.begin(9600);
  pinMode(up , INPUT);
  pinMode(down , INPUT);
  pinMode(enablePin1, OUTPUT);
  pinMode(dirPin1, OUTPUT);
  pinMode(stepPin1, OUTPUT);
  stepperZ = engine.stepperConnectToPin(stepPin1);
  stepperZ->setEnablePin(enablePin1, true); //nếu step hoạt động ở mức enable thấp thì gọi tham số là "true", step hoạt động ở mức enable cao thì gọi tham số là "false". setEnablePin() cho thư viện biết CÁCH bật / tắt trình điều khiển đầu ra của động cơ bước.
  stepperZ->setDirectionPin(dirPin1, true); //true = HIGH, false = LOW
  stepperZ->setAutoEnable(false); //true là bật, flase là tắt, setAutoEnable() cho thư viện biết khi nào bật/tắt trình điều khiển đầu ra của động cơ bước.
                                  //Nếu true, thì thư viện bước sẽ tự động kích hoạt các trình điều khiển đầu ra trước khi di chuyển và vô hiệu hóa chúng sau khi di chuyển. Thời gian chính xác được xác định bởi độ trễ bật/tắt. 
                                  //Nếu flase, thì thư viện bước không tự động bật/tắt trình điều khiển đầu ra và người lập trình ứng dụng phải bật/tắt rõ ràng với enableOutputs()/disableOutputs(). 
                                  //Nếu là enableOutputs() sẽ trả về cùng giá trị đặt của lệnh stepperZ->setEnablePin, nếu là disableOutputs() sẽ trả về giá trị ngược lại của lệnh stepperZ->setEnablePin                           
  stepperZ->enableOutputs();
}
//...............................................
void loop() {
  if ((digitalRead(up) == 0)&&(stepperZ->isRunning() == false)) {
    trucZup();
  }
  if ((digitalRead(up) == 1)&&(stepperZ->isRunning() == true)) {
    stepperZ->forceStopAndNewPosition(0);
  }
  if ((digitalRead(down) == 0)&&(stepperZ->isRunning() == false)) {
    trucZdown();
  }
  if ((digitalRead(down) == 1)&&(stepperZ->isRunning() == true)) {
    stepperZ->forceStopAndNewPosition(0);
  }
//...............................................
void trucZup() {
  if (stepperZ) {
     stepperZ->setSpeedInHz(800);       // steps/s
     stepperZ->setAcceleration(8000);    // steps/s²
     stepperZ->runForward();
  }
}
//...............................................
void trucZdown() {
  if (stepperZ) {
     stepperZ->setSpeedInHz(800);       // steps/s
     stepperZ->setAcceleration(8000);    // steps/s²
     stepperZ->runBackward();
  }
}

quangbinhhvac, 2022-01-11:
and I also can't stop the motor with command “stepperZ->forceStopAndNewPosition(0);”

gin66, 2022-01-11:
I assume, that up and down are two independent end position switches.
This would mean, either `digitalRead(up) == 1` or `digitalRead(down) == 1` is true, but not both.
Consequently there is always one of `digitalRead(up) == 0` or `digitalRead(down) == 0` true.
So in case one of the two `forceStopAndNewPosition()` are executed, in the next `loop()` run, the stepper will be started again.

quangbinhhvac, 2022-01-12:
Hi gin66,
my program is that when the up and down buttons are pressed, the motor will rotate in opposite directions.  When the button is not pressed the motor will stop.  I can't stop the engine
my 2nd problem is when i run low speed with micro 1/4, the motor is quite noisy, not smooth

quangbinhhvac, 2022-01-12:
Thank you very much, the program ran properly. But running at low speed 1000Hz and microstep 1/4, the motor is still quite loud. Maybe I need to do more research

closed 2022-01-12 by gin66