gin66/FastAccelStepper #106: How to use this ob CPU1?

apvlv, 2022-01-14:
Hello and thank you for this library. I want to use it with bluepad32 (http://gitlab.com/ricardoquesada/bluepad32) with esp32-idf to control the motor movements over a gamepad (Steelseries Nimbus). Though I'm getting errors when I call the init() function and my ESP32 WROOM-32 is getting an endless reboot loop. Apparently there is an issue with cpu cores and the developer of the library recommendef to try to use your stepper library on CPU1. Can you point me in the right direction how to do this?

Thank you very much.

apvlv, 2022-01-14:
```
E (451211) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (451211) task_wdt:  - IDLE (CPU 0)
E (451211) task_wdt: Tasks currently running:
E (451211) task_wdt: CPU 0: StepperTask
E (451211) task_wdt: CPU 1: IDLE
E (451211) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x40102A12:0x3FFBE7500x40082EBD:0x3FFBE770 0x40085BED:0x3FFDC210 0x40093915:0x3FFDC230 0x400D8356:0x3FFDC250 0x400946A5:0x3FFDC270 
0x40102a12: task_wdt_isr at /Users/xxx/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40082ebd: _xt_lowint1 at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40085bed: esp_crosscore_int_send_yield at /Users/xxx/esp/esp-idf/components/esp_system/crosscore_int.c:144

0x40093915: vTaskDelay at /Users/xxx/esp/esp-idf/components/freertos/tasks.c:1596

0x400d8356: StepperTask(void*) at /Users/xxx/esp/pan-tilt-head/build/../components/fast_accel_stepper/src/FastAccelStepper.cpp:69 (discriminator 1)

0x400946a5: vPortTaskWrapper at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/port.c:131

E (451211) task_wdt: Print CPU 1 backtrace

Backtrace:0x40085B81:0x3FFBED500x40082EBD:0x3FFBED70 0x4000BFED:0x3FFD2800 0x40094956:0x3FFD2810 0x40102CD7:0x3FFD2830 0x40102CE3:0x3FFD2860 0x400D3A5D:0x3FFD2880 0x40092C11:0x3FFD28A0 0x400946A5:0x3FFD28C0 
0x40085b81: esp_crosscore_isr at /Users/xxx/esp/esp-idf/components/esp_system/crosscore_int.c:92

0x40082ebd: _xt_lowint1 at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x40094956: vPortClearInterruptMaskFromISR at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/port.c:319

0x40102cd7: esp_task_wdt_reset at /Users/xxx/esp/esp-idf/components/esp_system/task_wdt.c:330

0x40102ce3: idle_hook_cb at /Users/xxx/esp/esp-idf/components/esp_system/task_wdt.c:80

0x400d3a5d: esp_vApplicationIdleHook at /Users/xxx/esp/esp-idf/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x40092c11: prvIdleTask at /Users/xxx/esp/esp-idf/components/freertos/tasks.c:3974 (discriminator 1)

0x400946a5: vPortTaskWrapper at /Users/xxx/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

https://github.com/ricardoquesada/bluepad32/issues/16

gin66, 2022-01-14:
The StepperTask is created with:
`xTaskCreate(StepperTask, "StepperTask", STACK_SIZE, this, PRIORITY, NULL);`

If you need to pin to a core, then you can change this line to:
`xTaskCreatePinnedToCore(StepperTask, "StepperTask", STACK_SIZE, this, PRIORITY, NULL,CPU1);`

If this works, then I could overload the init() function to allow setting the CPU core

apvlv, 2022-01-15:
Hello @gin66,

thanks to your help I could solve my issue and it's now working fine. I had to modify two lines of code:

like you said `xTaskCreatePinnedToCore(StepperTask, "StepperTask", STACK_SIZE, this, PRIORITY, NULL,1);`

and additionally I had to increase the vTaskDelay to 10ms to get it working: `#define TASK_DELAY_4MS 10 `

Thank you very much!!

gin66, 2022-01-15:
My guess is, that increasing the TASK_DELAY_4MS to 10 only reduces the likelihood of the problem…

So I would prefer to understand the root cause. 

Anyway, if you have a working solution, why not ?

Allowing to pin the task to the core makes sense to be added to the API.

apvlv, 2022-01-15:
Yes, adding it to the API would be awesome. As my understanding of threads and cores on the ESP32 is very limited yet, I have yet to find out what the root cause is. The vtaskdelay soulution was more or less just guessed... ^^ It seems to work fine now as I can say for an hour of trying it out. I post a solution if I find out more.

Thank you for the fast help!

gin66, 2022-01-15:
Could you please try on more thing. Just before the `vTaskDelay()`, add a `esp_task_wdt_reset();` And add at top: `#include <esp_task_wdt.h>`

gin66, 2022-01-15:
And what is the value of portTICK_PERIOD_MS ?

gin66, 2022-01-15:
The master branch contains the mentioned changes and one more (in case portTICK_PERIOD_ms > 4).
So it should be easier for you to check the effectiveness.

apvlv, 2022-01-15:
top, I can check it tomorrow and get back to you then! 👍

gin66, 2022-01-27:
no news are good news :-)

closed 2022-01-27 by gin66