gin66/FastAccelStepper #108: i need to output a frequency level of 100khz

quangbinhhvac, 2022-01-19:
Hi gin66,
i need to output a frequency level of 100khz for stepper to work with 1/16 micro.  but the library is only 25khz

gin66, 2022-01-19:
The stm32 is not supported from this library unless someone can provide a patch.

ST advertises [support for steppers](https://www.st.com/content/st_com/en/ecosystems/stm32-motor-control-ecosystem.html). Anything useful for you ?

quangbinhhvac, 2022-01-19:
yes i know about STM32CubeMX.  but I'm self-taught and not a programmer so using STM32CubeMX is quite a bit more difficult than Arduino IDE

quangbinhhvac, 2022-01-19:
i need to output a frequency level of 100khz for stepper to work with 1/16 micro. but the library is only 25khz

gin66, 2022-01-19:
The stm32 is not supported from this library unless someone can provide a patch.

closed 2022-01-19 by gin66

quangbinhhvac, 2022-01-19:
you don't understand what i mean, i mean when i use ATmega328p and want to output pulse with frequency 100khz to control motor with micro 1/16.  Because with the current 25khz frequency I can only put 1/4 . micro