gin66/FastAccelStepper #113: 0.25.0 does not compile

kthod861, 2022-01-28:
Hello,

After updating from 0.24.2 > 0.25.0, compilation can't be achieved.
Rolled back to 24.2 and it worked again.

Not sure copy/pasting the error will help due to its lenght but don't hesitate if you need it.

gin66, 2022-01-28:
Hi,

this is surprising. All architectures compile as part of github actions. Please share the error log.

gin66, 2022-01-28:
Should be ok now. Please try again

kthod861, 2022-01-28:
0.25 is gone from the update list...don't know if there suppose to be a new one.
Will wait a little bit, probably a delay before it beiing available...if not i'll just grab from github

kthod861, 2022-01-29:
0.25.2 don't trigger errors but motors don't run at all anymore. 

0.25.2 : no motors mouvements
0.25.1 : no motors mouvements
0.25.0 : compile error
0.24.2 : work flawlessly

i can try to prepare a unit test under 25.2 if you want with some prints here and there.

gin66, 2022-01-29:
Yeah. I cannot test SAMD, but had to rework the code for general change.
Now I have poked around some more, which most likely does not work.
Anyway, please give it a try.

@clazarowitz Sorry for poking around in your code, but have used the opportunity to clean up some unused variables/flags.

kthod861, 2022-01-29:
i was amazed to see that your example was working so i spent some time trying to understand the differences between it and mine.

Ended up pinpointing that if the first frame have any step value bellow 40 the motor won't run .
If i put a small value anywhere else it work but i think steps are lost in that case.

so from the example ino, juste replacing the framelist by

  uint16_t lframes[] = {30, 107, 107, 107, 107, 107, 107, 107, 107, 107,
                        107, 107, 107, 107, 107, 107, 107, 107, 107, 107,
                        107, 107, 107, 107, 107, 107, 107, 107, 107, 97};

is enough for the motor not to turn

It seems that adding some DelayToEnable help a bit for the example but still my main project does not work anymore (DelayToEnable was already there) .

gin66, 2022-01-29:
Thanks for finding reproducible issue. Just tried it here.
With avr simulator works as expected.
With esp32 no steps are generated.
Curious, what the root cause is.

gin66, 2022-01-29:
For this specific issue (actually a regression) I have added another HW test case.
With that test I have (hopefully) checked, that the applied fix is successful.
With this have released 0.25.3

Please check again

kthod861, 2022-01-29:
Quickly tested and it sounds right now, motors behave as before, thanks !

closed 2022-01-30 by gin66