gin66/FastAccelStepper #114: First instanced stepper gets stuck at 1khz 50% PWM

hnnswldschtz, 2022-01-30:
Hi

I’m using your fabulous library for a project. The control and calculations are awesome! but I run into a kinda volatile, but all the times happening problem

I’m running a WT32-ETH01 ethernet module which is build with a WT32-S1 which is a ESP32.
Two steppers are controlled from the controller trough 8825 stepper drivers to drive a XY moving head alike device. 
The controller receives its commands at any given time live via ethernet or WIFI as UDP transmitted OSC packets. Mainly X/Y pairs and speed/accel values. 
Its an PIO Arduino framework Project, including Wifi/udp/OSC libraries. 

Here is the problem: 
The Library works perfect regarding positioning, ramps, and the recalculations done with dir/speed/accel updates while on the move. But then, at completely random times (mostly after 3-5 mins): 

* stepper 1 gets stuck forever, always at a PWM of nearly 50% at a wavelength of 1ms (1 kHz). 
* the attached direction Pin stays stuck at its last state, no matter of incoming commands. 
* Everything else in the code continues to work: receive of OSC values, controlled of 2nd Stepper, dbg messages… so it looks like the channel is stuck in firing this signal forever at a lower level of the code. 
I can’t reproduce the trigger event but it will definitely happen in my setup. 

I spent quite a while on debugging and striping off every piece of code to pin down the problem. 
And somehow I thought found the source without really being able to understand it:
when I interchange the stepPin channels of stepper 1 and 2, its always the first Stepper that gets stuck. The one instanced at first (with following stepperConnectToPin() and enableOutputs() calls). 

So I thought I found a rather brutal workaround by setting up an unused FAS stepper instance before the instances I really want to use, including StepperConnectToPin() and enableOutputs() of that fake stepper instance. Like this both Steppers (now the 2nd and 3rd) ran quite long without problems, but after some hour long test, the stepper with the lower order number (now the 2nd) got stuck with the similar behaviour (PWM 50%, 1kHz). No Luck. 

I thought that could be interesting for you. My skills are not that evolved, so please excuse my rather superficial analysis.I tried to play around with queue2mapping struct its references in StepperISR_esp32.cpp but that didn’t work at all. Actually when it goes that deep it seems to be above my paygrade. 

I didn’t make a minimal version so far, but I will, if this seems to be something of interest for you. 
From my understanding it might be a MCPWM/Timer/PCINT problem. Of course this happens somehow because the other libraries used (OSC/UDP) in the code.

gin66, 2022-01-30:
Sounds like an awesome project, which makes use of the great connectivity of esp32. 

As a start for analyzing the root cause, here a couple of questions:
- which library version are you using ? 
- Is this preprocessor variable defined: MCPWM_TIMER0_PHASE_DIRECTION_S ?
- and what about ARDUINO_ARCH_ESP32 ?

gin66, 2022-01-30:
Could you please add these lines, which outputs the pulse counter's current and limit value e.g. once every 10s or so ?Serial.print("PCNT: ");
Serial.print(PCNT.cnt_unit[(pcnt_unit_t)0].cnt_val);
Serial.print(" ");
Serial.println(PCNT.conf_unit[0].conf2.cnt_h_lim);

Or send the value via UDP, or another method.

I am wondering, if in this status the pulse counter is still counting, or which value it is frozen at. And if the lim_h value is perhaps zero. The latter may be the probable root cause.

gin66, 2022-01-30:
After adding the prints about the PCNT value, please check the latest master branch. I have added an experimental fix for the case that lim_h value has been set to 0. The problem is, that this special case should be very unlikely, or not happen at all. So it may not be related and the fix itself may be wrong. Anyway, has you have a stable failing system, it is worth a try.

If addition: You could add a second pulse counter and watch the pulse counting from an independent module using once `stepper->attachToPulseCounter(7)` and then repeatedly output the result of `stepper->readPulseCounter()`.
Especially if the experimental patch is of help, then the second pulse counter can help to identify miscounting of steps.

hnnswldschtz, 2022-01-31:
Hi gin66,
thanks a lot for sharing your time and skills! 
To answer all your questions:
• I was using 0.24.0
• ARDUINO_ARCH_ESP32 is defined
• MCPWM_TIMER0_PHASE_DIRECTION_S is NOT defined

But that doesn’t matter anymore, changing to your latest version makes it run perfectly!
I wasn’t aware of your new version. 
Out of interest, i compiled my code using 24.2 and 25.0. the error disappears with 25.0. and later.  

And maybe for further analysis: at FAS v24.xx the suggested stepper->attachToPulseCounter(7)  / stepper->readPulseCounter() approach showed, that once the stepper got stuck, the counter decremented until the it overflowed at -2^14 restarting then from zero. Never stopped, repeating forever.

Thanks a lot for your help and this great library!

closed 2022-02-06 by gin66