gin66/FastAccelStepper #115: forwardStep unwanted delay

alba-ado, 2022-01-31:
Hello, I've noticed that you added a delay in the `performOneStep(bool count_up, bool blocking)` function with the usage of `yield_if_supported();` However, it significantly slows down the stepping motion of the motor when used in a loop. I think instead of using `vTaskDelay(1)` you can alternatively use the `yield()` call in order to yield to higher priority tasks. This way, you won't be blocking any tasks and the motor runs faster. If anyone wanted to use a delay, they can use the non-blocking version and implement a while loop themselves.

gin66, 2022-01-31:
Thanks for the proposal. I have not known the `yield()` function before. 
I have included this in the 0.25.4, which is already released to github.

What I do not understand actually, is the relation to "significantly slowing down the stepping motion". The blocking means, that it will only return, after the step has been done. Consequently it has to busy-wait. The yield() and the vTaskDelay() only tells the OS, that other tasks can be processed while waiting. In this regard, perhaps the vTaskDelay() may even better, because lower priority tasks will perhaps run, which for yield() I am not sure. And the vTaskDelay() - and I haven't looked up the docs - should push the task to sleep and let others run. But still, why is this impacting the stepping motion ?

devrim-oguz, 2022-01-31:
Yes, I don't know what is your intended use case for this function in this library, but I am using it for a scissor mechanism to go forward until it touches an object. Fot that regard, I can't use the moveTo function since I need really fine control but I've been calling this function inside a while loop with hunderds of microseconds of delay. After I updated my library, I noticed the motor slowing down greately. So, I tracked this issue until here and replaced the vTaskDelay with yield. It works better but yield can behave quite differently sometimes. So it is better to leave it like the old version where it gets stuck in the loop. As I stated in another issue, sepersting the functionality would be best to cover all usage cases, since I was able to use vTaskDelay whenever I wanted to after making the motor step or using the unblocking version of the forwardStep. But it is up to you to implement this or not. I just changed my version of the library.

gin66, 2022-01-31:
That's the funny thing, I do not have a use case for this function. It was just added as request from you. If the yield() does not work well for you, then I can remove it again. No problem

devrim-oguz, 2022-01-31:
Oh my god, thanks man! 😊
I'm your number one fan hahaha

devrim-oguz, 2022-01-31:
[image]

I'm using your library for this mechanism and some other things.

gin66, 2022-01-31:
Thanks for the picture. Always nice to see the application for what the lib is used for.

The v0.25.5 has the yield() removed and old behavior is back. Sorry for the inconvenience