gin66/FastAccelStepper #117: Compilation fails for ESP32 Dev on Arduino IDE

jay2310a, 2022-02-10:
Do I do anything absolutely wrong?

With this Error:

> Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, Core 1, Core 0, None"

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp: In function 'void apply_command(StepperQueue*, bool, uint32_t*)':

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:61:35: warning: 'esp_err_t rmt_memory_rw_rst(rmt_channel_t)' is deprecated: use rmt_tx_memory_reset or rmt_rx_memory_reset instead [-Wdeprecated-declarations]

       rmt_memory_rw_rst(q->channel);

                                   ^

In file included from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\common.h:107,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\FastAccelStepper.h:5,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR.h:3,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:1:

C:\Users\jahi\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:898:11: note: declared here

 esp_err_t rmt_memory_rw_rst(rmt_channel_t channel)

           ^~~~~~~~~~~~~~~~~

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:61:35: warning: 'esp_err_t rmt_memory_rw_rst(rmt_channel_t)' is deprecated: use rmt_tx_memory_reset or rmt_rx_memory_reset instead [-Wdeprecated-declarations]

       rmt_memory_rw_rst(q->channel);

                                   ^

In file included from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\common.h:107,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\FastAccelStepper.h:5,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR.h:3,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:1:

C:\Users\jahi\Documents\ArduinoData\packages\esp32\hardware\esp32\2.0.2/tools/sdk/esp32/include/driver/include/driver/rmt.h:898:11: note: declared here

 esp_err_t rmt_memory_rw_rst(rmt_channel_t channel)

           ^~~~~~~~~~~~~~~~~

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp: In function 'void tx_intr_handler(void*)':

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:180:14: error: 'RMT_CH0_TX_END_INT_ST' was not declared in this scope

   if (mask & RMT_CH ## ch ## _TX_END_INT_ST) { \

              ^~~~~~

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:193:3: note: in expansion of macro 'PROCESS_CHANNEL'

   PROCESS_CHANNEL(0);

   ^~~~~~~~~~~~~~~

In file included from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\FastAccelStepper.h:5,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR.h:3,

                 from C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:1:

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\common.h:113:43: error: 'RMT_CHANNEL_MEM' was not declared in this scope

[... 1021 lines cut ...]

   uint32_t *data = FAS_RMT_MEM(channel)+63;

                    ^~~~~~~~~~~

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\common.h:113:43: note: suggested alternative: 'RMT_CHANNEL_MAX'

 #define FAS_RMT_MEM(channel) ((uint32_t *)RMT_CHANNEL_MEM(channel))

                                           ^~~~~~~~~~~~~~~

C:\Users\jahi\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:352:20: note: in expansion of macro 'FAS_RMT_MEM'

   uint32_t *data = FAS_RMT_MEM(channel)+63;

                    ^~~~~~~~~~~

exit status 1

Fehler beim Kompilieren für das Board ESP32 Dev Module.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.
>

gin66, 2022-02-10:
Apparently arduino 1.8.19 uses a more recent version of the arduino-esp32 component than platformio, which has introduced incompatible changes.

Need to update the lib. Hope this weekend can do it.

jay2310a, 2022-02-10:
I fixed the issue by adding includes in common.h.

I am not sure how the rmt_reg.h definitions used to be in earlier releases, but this compiles at least...

#if CONFIG_IDF_TARGET_ESP32

...

#define SUPPORT_ESP32_RMT
#include <driver/rmt.h>

**#include <driver/periph_ctrl.h>
#include <soc/rmt_periph.h>
#include <soc/rmt_reg.h>
#include <soc/rmt_struct.h>**

gin66, 2022-02-10:
Thanks for investigation. I still need to check, if this is backwards compatible.

With these changes I expect, that there are still the depreciation warnings. Correct ?

gin66, 2022-02-11:
Should be ok now with 0.27.5
Thanks for the hint with the include