gin66/FastAccelStepper #118: Different acceleration and deceleration

ninamercedez, 2022-02-14:
Is there a possibility to have a different acceleration and deceleration in one movement.
e.g. moving 100 mm with an acceleration of 3000 mm/s² and decelerate with 10000 mm/s² ?

gin66, 2022-02-15:
The current ramp generator does not support this. Eventually there will be a new one, which can support this feature, but there is no plan to start on implementation  in the coming weeks.

From application side, you can try this approach:
```
void loop() {
   if (!stepper->isRunning()) {
      stepper->setSpeed(500);
      stepper->setAcceleration(3000);
      stepper->moveTo(10000 - stepper->getCurrentPosition());
   }
   if ((stepper->rampState() & RAMP_STATE_MASK) == RAMP_STATE_COAST) {
       stepper->setAcceleration(10000);
       stepper->applySpeedAcceleration();
   }
}
```
Hope this compiles. The stepper will run back and forth between position 0 and 5000. If in your app, the ramp does not reach the max. speed, then the proposed code will not be able to set the deceleration value.

JanBluePix, 2023-06-10:
Entering a new value for a acceleration as deceleration after the acceleration is done works as described. Setting a separate value for acceleration and deceleration would be nicer though.

gin66, 2025-08-10:
Not sure if monitoring the state for coasting is too complicated. Sure, it is more convenient to have this as an option.

A comparably straightforward implementation is to reduce the length of the deceleration ramp by a factor of 2, 4, 8, 16, or more. On AVR, division is slow, so I prefer to use only a shift operation because that calculation needs to be done on every step during acceleration.

A free choice of deceleration factor would mean recalculating the length of the deceleration ramp for every acceleration step, too.

AdamDrefs, 2025-10-21:
I agree that it would be nice to have separate acceleration and deceleration profiles, but I agree with gin66 on this one, since it takes time to implement and would be mostly unnecessary given we already have a solution. I might take some time to see if I can make a fork that implements it, but that would involve figuring out how the heck this thing works and how to manipulate the internal code, so it'll take a long time if I ever do it.

fray-d, 2026-05-09:
+1 on this feature.

The suggestion above to check for coasting only works if the acceleration is fast enough to reach max speed prior to the midpoint of the move, otherwise the ramp generator will immediately decelerate (triangular motion profile). With a true implementation of separate accelerations you could accelerate slowly with both the maximum speed and a sudden halt much closer to the end of the move. Likewise you could accelerate quickly and then decelerate slowly without risk of overshooting the endpoint.