gin66/FastAccelStepper #119: Benefit of mcpwm?

cranefist, 2022-02-23:
Switched to this library, and trying to figure out how this mcpwm affects me in practical terms.

Does the mcpwm on ESP32 generate the steps? Meaning that the CPU does not have to use time for the step generation?
Is it truly non blocking? Much like using VACTUAL on TMC2209 for example? Where the driver generates the steps?

Im wondering do i still need to allocate the same resources for the stepper as before or does this mcpwm reduce the load on the CPU.

gin66, 2022-02-23:
The mcpwm is generating the steps. If the stepper is running at 200 kHz, then normally the mcpwm is running freely at that frequency and the pulse counter is counting the steps. At 200 kHz the pcnt will count up to ~200-255 and only then will create ONE interrupt to the CPU. During generation of the 200-255 steps, the CPU is not loaded.

In order to let the mcpwm/pulse counter know, what to do, there is still a OS-task running in parallel. The main application is not involved in this background task and free to do other operations, while the stepper is running. So the calls to FastAccelStepper are non-blocking (unless the block parameter is set to true, for those API-calls having this kind of support).

No idea, what VACTUAL is...

In a nutshell, the load for stepper operation on the esp32 is reasonable. I haven't measured this, but I would expect something in the single digit percentage range.

cranefist, 2022-02-23:
Thanks a lot for this clarification. Seems ESP32 is even more awesome than i thought previously. Going to switch to this library on all my projects. Seems you fixed the compiling errors it had on ESP32, thanks for that.

VACTUAL is a command for TMC stepper drivers that you can use in UART mode, where the driver chip generates the steps. It just sets the speed and the stepper runs as long as you stop it (by setting the speed to 0), but its unable to count steps.

If you ever plan to add closed loop support to this, it might be a interesting feature.
Close loop support would overall be highly appreciated. For encoders like AS5600 that are really cheaply available and easy to adapt.

gin66, 2022-02-23:
Thanks for the feedback in comparison to FlexyStepper. 

Sorry to give you bad News: There is no plan to add encoder support and I would not even integrate any pull request with an implementation of this kind of feature.

cranefist, 2022-02-23:
I seem to have a weird issue, i switched two projects into this library. Bot have really similar code & hardware.

But on one of these the motor only moves forward. It moves the correct amount, but always forward.

Everything checks out, i get perfectly correct data from targetPos()) & getCurrentPosition() and from getDirectionPin()

And when i tested a simple code that has only the stepper motor code, it works and moves in both directions. But even while using the whole code,  even if i put this into the setup before any of my code even runs:

  stepper->moveTo(300,true);
  delay(3000);
  stepper->moveTo(0,true);
  delay(3000);

It moves 300 forward on both of these instances.

This is my setup:

engine.init();
stepper = engine.stepperConnectToPin(STEP_PIN);
stepper->setDirectionPin(DIR_PIN);
stepper->setAutoEnable(false);
stepper->setSpeedInHz(100); 
stepper->setAcceleration(100);

Does anything come to mind that might cause this?

cranefist, 2022-02-23:
My dreams are crushed, hope is lost.

gin66, 2022-02-24:
yeah. Life is hard

closed 2022-02-24 by gin66