gin66/FastAccelStepper #120: setExternalEnableCall

cranefist, 2022-02-24:
Im having difficulties understanding this, and could not find any examples about using it.

As i'm controlling the stepper driver via UART,  i turn it ON via driver.toff(3) amd OFF via driver.toff(0) .

Do i still need to define an enable pin for it to work?

I really have no clue how this is supposed to be set up, i will create a bool function that executes driver.toff(3) on FALSE and driver.toff(0) on TRUE?

gin66, 2022-02-24:
The enable pin is not mandatorily driven by FastAccelStepper. The application can do the same, if needed or special requirements are there.
 
Anyway, there is no example for the callback, because I do not use it. It was added on [user request](https://github.com/gin66/FastAccelStepper/issues/14)

In your `setup()` just write:
```cpp
   stepper->setEnablePin(100, false); // use just 100 as number, false means high active
   stepper->setExternalEnableCall(my_enable_callback);
   stepper->setAutoEnable(true);
   stepper->setDelayToDisable(1000); // 1s off delay
```

and you need the callback function:
```cpp
bool my_enable_callback(uint8_t enablePin, uint8_t value) {
   // enablePin == 100 in this example, if you have several steppers with same callback,
   // then value of enablePin can be used to enable/disable the appropriate stepper
   if (value == HIGH) {
      driver.toff(3);
   }
  else {
      driver.toff(0);
  }
  return value;
}
```

I have not compiled or tested this proposal. Just from theory, this should work.

cranefist, 2022-02-26:
Thanks a lot for this.

I did a quick test, but im unsure did it work. I will test it more thoroughly tomorrow, and will report back.

I had my own code doing the same, but if this works its just tidier.

cranefist, 2022-02-27:
I tested it with this simple code, it does not seem to work.. but is this because i'm using a blocking delay?
```
//STEPPER TMC2209
#include <TMCStepper.h>

#define DIR_PIN          5 // Direction
#define STEP_PIN         18 // Step

#define STALL_VALUE     255 // [0..255]
#define SERIAL_PORT Serial2 // HardwareSerial port
#define DRIVER_ADDRESS 0b00 // TMC2209 Driver address according to MS1 and MS2
#define R_SENSE       0.11f // 0.11 for MKS TMC2209
#define AMPS            220 // MAX CURRENT 1700

TMC2209Stepper driver(&SERIAL_PORT, R_SENSE, DRIVER_ADDRESS);
using namespace TMC2209_n;

#include "FastAccelStepper.h"
FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {

Serial.begin(115200);
SERIAL_PORT.begin(115200,SERIAL_8N1, 16, 17);      // INITIALIZE SERIAL2 UART FOR TMC2209
delay(1000);
  
//TMCSTEPPER SETTINGS
driver.begin();                                               
driver.toff(3);                  // enable driver in software
driver.rms_current(AMPS, 0.01);  // motor RMS current "rms_current will by default set ihold to 50% of irun but you can set your own ratio with additional second argument; rms_current(1000, 0.3)."
//driver.seimin(1);              // minimum current for smart current control 0: 1/2 of IRUN 1: 1/4 of IRUN
driver.semin(3);                 // [0... 15] If the StallGuard4 result falls below SEMIN*32, the motor current becomes increased to reduce motor load angle.
driver.semax(1);                 // [0... 15]  If the StallGuard4 result is equal to or above (SEMIN+SEMAX+1)*32, the motor current becomes decreased to save energy.
driver.sedn(4);                  // current down step speed 0-11%
driver.seup(2);                  // Current increment steps per measured StallGuard2 value 5 seup0 %00 … %11: 1, 2, 4, 8
driver.TPWMTHRS(0);              //When the velocity exceeds the limit set by TPWMTHRS, the driver switches to spreadCycle.
driver.blank_time(24);
driver.TCOOLTHRS(0xFFFFF);       // FFFFF
driver.microsteps(0);        // microsteps
driver.pwm_autoscale(true);      // Needed for stealthChop
driver.en_spreadCycle(true);     // spreadCycle
driver.shaft(false);             // direction
//driver.ihold(2);               // hold current  0=1/32 … 31=32/32
driver.iholddelay(3);            // 0 - 15 smooth current drop
//driver.irun(31);  
driver.SGTHRS(STALL_VALUE);
//driver.I_scale_analog(0);   // if 5v vdd

//FAST ACCEL STEPPER
engine.init();
stepper = engine.stepperConnectToPin(STEP_PIN);
stepper->setDirectionPin(DIR_PIN);
stepper->setEnablePin(100, false);
stepper->setAutoEnable(true);
stepper->setDelayToEnable(0);
stepper->setDelayToDisable(1000);
stepper->setSpeedInHz(250);  // STEPS PER SECOND
stepper->setAcceleration(1000);
stepper->clearPulseCounter();
stepper->setCurrentPosition(0);
}

void loop() {
Serial.println(stepper->targetPos());
Serial.println(stepper->getCurrentPosition());
Serial.print(driver.cs2rms(driver.cs_actual()),DEC);Serial.println(" MA MOTOR CURRENT"); 
Serial.print(driver.SG_RESULT());Serial.println(" STALLGUARD"); 
Serial.print(driver.microsteps());Serial.println(F(" MICROSTEPS"));
  stepper->moveTo(600,false);
  delay(10000);
  stepper->moveTo(200,false);
  delay(10000);

}

//TMC2209 ON/OFF--------------------------------------------
bool enableTMC(uint8_t enablePin, uint8_t value) {
   // enablePin == 100 in this example, if you have several steppers with same callback,
   // then value of enablePin can be used to enable/disable the appropriate stepper
   if (value == HIGH) {
      driver.toff(3);
   }
  else {
      driver.toff(0);
  }
  return value;
}
```

cranefist, 2022-02-27:
I think i figured it out, i disabled my debugging serial print task. So the problem was that i was calling ```stepper->enableOutputs()``` in one of my serial outputs. Maybe i misunderstood what it does, i thought it was just to see if the driver was enabled or not.

gin66, 2022-02-27:
Happy to hear, that it works. So the issue can be closed

closed 2022-02-27 by gin66