gin66/FastAccelStepper #121: Hardware for high speeds and accelerations in multi-axis configuration

DSMaryam, 2022-02-28:
Hi !
First, thank you for the great work you have produced here. 

I would like to use a compact hardware configuration for multi axis control (2 motors or 3). Currently, the best I have is an Arduino Portenta equiped with STM32H747, but I understand STM is not covered by the library ; is there a way it could be render compatible ?

Or, rather than Portenta, what would be my best options to get multi axis high speeds and accelerations in a compact arduino, or other hardware, using FastAccelStepper ?
For comparison, the performances I get with one motor on atMega32U4 are right just sufficient for my application.

Thank you in advance !

DSMaryam, 2022-03-01:
Many thanks for this detailed response ! 

The big problem with ST stepper motor drivers is that, for some reasons, it has become difficult to order them (distributors' stocks are low) and I would be very annoyed to start working with a technology and not be able to order it anymore in the future. Plus, adding my compact constraint reduces the choice on drivers. Same goes for Trinamic drivers that can also be very powerful. I thought I would never have this problem with Arduinos.

I will definitely try using my Portenta with AccelStepper.

To be more precise, high speeds and accelerations for me are at the order of 100k microsteps/s and 1M microsteps/s^2. I will probably go on and try ESP32, for the fun of being impressed, although there may also be the ordering issue ;)

gin66, 2022-03-01:
100kSteps/s will be difficult for AccelStepper.
FastAccelStepper supports 100kSteps/s only with esp32. Best to get this.

closed 2022-03-02 by DSMaryam