gin66/FastAccelStepper #126: I want to change the speed limit "step/s"

quangbinhhvac, 2022-04-16:
Hi gin66,
I want to change the speed limit "step/s"
in old versions i found it in "FastAccelStepper.h". I can't find it in the new version
please help me

gin66, 2022-04-16:
Hi,

hope this answers your question as there are four different calls. For step/s, the call `setSpeedInHz(uint32_t speed_hz)` is adequate. 1/s = 1Hz.
```
  // Setting the speed can be done with the four `setSpeed...()` calls.
  // The new value will be used only after call of these functions:
  //
  // - `move()`
  // - `moveTo()`
  // - `runForward()`
  // - `runBackward()`
  // - `applySpeedAcceleration()`
  // - `moveByAcceleration()`
  //
  // Note: no update on `stopMove()`
  //
  // Returns 0 on success, or -1 on invalid value.
  // Invalid is faster than MaxSpeed or slower than ~250 Mio ticks/step.
  int8_t setSpeedInUs(uint32_t min_step_us);
  int8_t setSpeedInTicks(uint32_t min_step_ticks);
  int8_t setSpeedInHz(uint32_t speed_hz);
  int8_t setSpeedInMilliHz(uint32_t speed_mhz);

```

quangbinhhvac, 2022-04-17:
Hi,
with atmega328 i need a speed of 50 rpm/s and with microstep 3200. i need a pulse with frequency 160000Hz
in the old version of the library i found a limit change

gin66, 2022-04-17:
best to use an esp32. For atmega328, 160kHz is not supported

quangbinhhvac, 2022-04-19:
Hi,
what is the frequency (Hz) limit of atmega328?

gin66, 2022-04-21:
allows up to 76000 generated steps per second for single stepper operation, 37000 for dual stepper

quangbinhhvac, 2022-04-23:
Thank you so much

closed 2022-04-23 by gin66