gin66/FastAccelStepper #129: Is there a way to determine the time until stepper reaches the destination?

hnnswldschtz, 2022-05-02:
This is not an issue, rather a question how to approach a problem.
I'm using your really great library on an ESP32, it works perfect!
I have two steppers in an X/Y movinghead like setup where I want to run them in a smooth fashion, basically as slow as possible to reach their targets within a given time. Therefore I would like to compute upfront the total time until a stepper reaches its destination from its actual position including all needed ramps. Im stuck here a how to use the rampCalcuators so I would be able to reproduce your calculations and get a more or less approximation of the needed time. 
Could you give me a hint how I could tackle this?

hannes

gin66, 2022-05-02:
There are two resources, you could check out:
- [ramp doc](https://github.com/gin66/FastAccelStepper/blob/master/extras/doc/ramp.md)
- [test_seq_08](https://github.com/gin66/FastAccelStepper/blob/master/examples/StepperDemo/test_seq_08.cpp)

The latter may be difficult to understand, because the purpose of the test sequence is testing and not documenting....

gin66, 2022-05-04:
Setting a new position does not imply a change in magnitude of acceleration, but it can change the sign. So the absolute values of old a and new a are the same.

For your questions: The ramp generator always keeps track, how many steps (s_d) are needed to come to a standstill from the current position/speed. This is completely independent of the question, if the stepper is currently accelerating, coasting or decelerating. The ramp generator maintains s_d accordingly. If a new position is given, then there are three possible cases:
1.) The new position is in same direction and more steps away than s_d. So the stepper will continue any ongoing acceleration or coasting.
2.) The new position is in same direction exactly s_d steps away. So the stepper must immediately start deceleration ramp (unless deceleration is going on anyway) and will stop at point.
3.) The new position cannot be reached without overshoot. Consequently the stepper will immediately start decelerating and come s_d steps later to a halt. From this standstill, it is just a normal move running to the new position (with acceleration, eventually coasting and deceleration).

closed 2022-05-11 by gin66