gin66/FastAccelStepper #131: Limit position counter to specific range

mat02, 2022-05-15:
Hi!
My application requires continous motion of the motor and reporting current position (angle, to be specific) on external sync signal (I'm using Arduino Nano with ATmega328p). Is there a possibility to limit the position counter to some specific range, i.e. position reverts to minimum after some arbitraly chosen value? 
 For now , I'm calling setCurrentPosition in ISR every time getCurrentPosition reports value bigger than step count for 360 degree, but it would be nice if this counter could reset on it's own.
Best regards,
Mateusz

gin66, 2022-05-15:
This is quite a specific use case and there is no suitable feature implemented. On esp32, a free pulse counter could be attached and used accordingly. But this is a hw feature, which has no counterpart on the 328p.

An application alternative could be to not use getCurrentPosition() in an interrupt, but to maintain a reference position (16 or 32bit). The difference between the current position and the reference value could be interpreted as the needed limited value (modulo 360). Whenever this value is >= 360 (or once a while), the reference value is advanced in 360 degrees. This will work even in case of overflow. If this can be done in your application, then this would be my preferred choice over an interrupt solution.

mat02, 2022-05-16:
Hi gin66, thank you for prompt reply :)
The solution that you suggest is quite interesting, I will try it out :D
Thanks again :)

closed 2022-05-16 by mat02