gin66/FastAccelStepper #134: forceStop at the end of move/moveTo or setting different accelerating/decelerating values

ilovehotcakes, 2022-06-17:
Greetings,

I'm working on an ESP32 project and I'm utilizing your library from accelstepper library. It's been working great so far but the stepper motor slips when decelerating because I'm pulling on a spring. The stepper motor slips when it approaching the target position and slows down. I was wondering if there is a way to use forceStop when moveTo approaches the target position? Or is there a way to set different acceleration and deceleration values.

Thank you!
Jason

gin66, 2022-06-17:
Thanks for using FastAccelStepper. Unfortunately I have difficulty to understand your application problem with the spring causing the stepper to slip. Anyway, there is no mechanism available as you describe. It is up to the application to do the necessary control. Here are two proposals:

- If the stepper always reaches full speed, then you could change the acceleration to a very high value, which resembles to forceStop. Full speed can be detected by `(stepper->rampState() & RAMP_STATE_MASK) == RAMP_STATE_COAST`
- Otherwise the app can check the position periodically and initiate  forceStop() on reaching the targetPosition(). Here then can use runForward() to avoid deceleration by the ramp generator.

ilovehotcakes, 2022-06-17:
Thank you for the prompt reply. I'm trying to motorize my manual honeycomb shades. They have a "tape-like" spring mechanism connected to the axle to provide resistance and preventing the shades from closing by themselves. FastAccelStepper is really accurate at stopping when lowering (closing) the shades because it is pulling on the spring (motor is working against the spring). However, when the shade is rising (opening), the spring is pulling on the motor in the same direction that the motoring is moving and it is causing the motor to stop further than expected (overshooting) when decelerating. There is no problem when I use a large decelerating value but it is being complicated by the TMC2209 driver's StallGuard 4 feature and it requires a slow acceleration so it doesn't trip SG when the motor starts moving. Therefore, I was hoping that I could set the acceleration and deceleration values independently or specifying to use forceStop() when moveTo() approaches target position.

I have tried both of your proposals:

- Which API do I call to apply the acceleration speed without needing to call one of the move APIs?
- I have one core on the ESP32 constantly checking stepper->getCurrentPosition() but it's easy for the the stepper to move pass the target position and again causing overshoot.

gin66, 2022-06-17:
Thanks for your detailed explanation. Separate acceleration and deceleration values would be nice to have, but it would make the driver way more complicated and getting the corner cases right, would need much more time.

- Which API do I call to apply the acceleration speed without needing to call one of the move APIs?
   `setAcceleration(int32_t step_s_s)` followed by `applySpeedAcceleration();`
- I have one core on the ESP32 constantly checking stepper->getCurrentPosition() but it's easy for the the stepper to move pass the target position and again causing overshoot. 
  An alternative would be to use a free pulse counter connected to your stepper and use that info for determination, when to stop `forceStop()`. Just wondering, how fast is your stepper running ?

Can you increase the current through your stepper, so that the stepper has more power ?

BTW: I haven’t known Stallguard before, so I am not sure, how it works exactly and my following assumption could be completely wrong. Nevertheless here it is: I have the impression, that this specific load case, where the motor gets accelerated by the applied load, hasn’t been in the requirement spec for stallguard. It guards against STALL and not against speeding up by external force. Perhaps the wrong technology for your application ? Can this be turned off completely and then check for the result ?

ilovehotcakes, 2022-06-17:
Thank you for the reply!

- I see. I didn't call applySpeedAcceleration() after setAcceleration().
- Thanks. I will look into the free pulse counter. My stepper is running 8288 steps per second or 1 rev per second. (5.18:1 gearbox ratio * 8 microsteps * 200 steps per rev for stepper)

I have tried to experiment with more current through the stepper but it doesn't matter as much when it is decelerating in the same direction as the spring is pulling in. 

I'm still learning the details of StallGuard 4 but basically it monitors the mechanical resistance via the back EMF and stalls the motor if the back EMF exceeds the set limit. It is a "nice-to-have" feature since the motorized shades will work without it. I can use SG to set the minimum position automatically (it stops by itself when it reaches the top) or prevent the shades from extending beyond it's physically limits without harming the stepper motor. I believe the reason a fast acceleration when speeding up stalls the motor because the acceleration creates a strong mechanical resistive force and thus tripping SG. Setting the right values for SG to work properly is an art that I'm still learning. I can definitely turn off SG and set fast acceleration values and that's probably what I will have to do if I can't figure out the overshooting issue. Otherwise, your library and SG works beautifully! Thanks!

closed 2022-06-17 by gin66