gin66/FastAccelStepper #137: SAMD21 compatibility (Arduino zero)

danielcostanzi18, 2022-07-16:
Hi, I'm working on porting this library into a SAMD21 MCU (same as Arduino Zero), has anybody already tried it? Do you see any critical issue? I'm wondering so since it has not been implemented yet. Any suggestion on how to deal with this would be really appreciated.
Thank you, hope to contribute to this library :)

gin66, 2022-07-16:
A port to the SAMD21 would be great. AFAIK no one has worked on a port yet. Looking at the [data sheet](https://content.arduino.cc/assets/mkr-microchip_samd21_family_full_datasheet-ds40001882d.pdf), the SAMD21 has several possibilities to generate pulses using either TC or TCC (perhaps even DMAC). The AVR code could be used as basis for your port. TC and TCC will require one interrupt per step, so monitoring cpu load by the interrupts and fast code are essential for high speed operation.  If any advanced feature of TCC may be of help, I have not checked.

In general an oscilloscope is of great help. For either checking the generated pulses or debugging interrupt routines. If you have none, best to get a cheap digital one.

And yes, if this port project is neither of fun for you or in any respect mission critical to you, better to not try. It will take quite some time and debugging interrupts is not for everyone.  In case you will start, please keep me involved. Perhaps I can help.

Looking forward to your pull request :-)

danielcostanzi18, 2022-07-19:
Hi @gin66 , thanks for your reply. As you said this port would take quite a lot of time, I'm now considering other ways to control my stepper motors. Maybe in the end I will decide to port this library to SAMD21, we'll see! ;)

closed 2022-07-19 by gin66