gin66/FastAccelStepper #142: Direction pin optional?

cranefist, 2022-08-19:
Hello

Can you clarify this? Could i somehow use the UART connection to change direction instead of the direction pin in this library?

As TMCStepper library allows me the change the direction via : driver.shaft(false/true)

This would fee up one pin.

Or does it just mean, that the library can operate without the direction pin?

gin66, 2022-08-19:
if the stepper shall run forward and backward, then a direction pin is required.
The direction pin can be controlled by the application by any means (e.g. shift register or else). FastAccelStepper just need to know the external callback. Check #132 for reference

cranefist, 2022-08-28:
Sorry i did not notice your response.

I tested this, but it seems setExternalCallForPin does not exists?

`'class FastAccelStepper' has no member named 'setExternalCallForPin'`

gin66, 2022-08-30:
You are right. It should be `engine.setExternalCallForPin(exampleCallback);`

Your example callback looks good.

cranefist, 2022-08-30:
I actually tested that also, it did not work either. 
`undefined reference to `FastAccelStepperEngine::setExternalCallForPin(bool (*)(unsigned char, unsigned char))'`

gin66, 2022-08-30:
cool. The implementation was just missing. Having no test for a feature opens the door for failure…

Fix has been pushed to master. But still no test….

cranefist, 2022-08-30:
Ok good, now it compiles.

Sadly i just blew up my last driver, need to wait for a replacement before i can confirm will it actually work :)

I also moved to using attachToPulseCounter, it seemed to work fine. Do i benefit anything from it? Is it more efficient way of counting steps?

gin66, 2022-08-31:
attachToPulseCounter just gives you access to real-time ead of calling a lengthy API function. I for me it is great for verifying to have no pulse losses during development.

So I wait for your confirmation. Eventually I will update the driver to use those callbacks as internal standard. This will reduce complexity my removing a couple of `if(){}` structures.

cranefist, 2022-08-31:
I shall report back in about 1 month :) Developing stuff is fun when you order everything from china.

cranefist, 2022-09-26:
One month later, as promised, does not seem to work. Motor just stops and sets hold current when it should change direction.

gin66, 2022-09-26:
Pity.  So the implementation is faulty and I need to debug it. Hopefully I can do it over the weekend.

cranefist, 2022-09-27:
Let me know if you need me to test something, its worth implementing.

gin66, 2022-10-01:
I have tested with avr simulator and fixed a bug. Hopefully it works now for esp32, too. 
The version is 0.28.0

cranefist, 2022-10-01:
It works, but it seems to be unstable.. sometimes it does not change the direction. And it seems to be really random when it suddenly does not change the direction.

gin66, 2022-10-02:
Thanks for checking. I think, I see the issue in the esp32 code using mcpwm/pcnt. need to figure out a solution.
The esp32 rmt looks better.

gin66, 2022-10-02:
github automatism is surprising….

Anyway, I hope the error has been fixed.

cranefist, 2022-10-03:
Still the same problem. I placed a toggling boolean in the function, and it does do what its supposed to do. 

Seems the UART command is not being registered always, i need to confirm that is been switched...

If i understood correctly, if i return the opposite value in the function it then repeats it again? Until it returns the correct value?

gin66, 2022-10-03:
yes, if the opposite  value is returned, the driver will try again. The brute force method is problematic, if you run more than one motor, because this may cause another motor run out of stepper commands. So instead of the while loop, perhaps this would work:
```
bool DirectionChange(uint8_t pin, uint8_t value) {
   driver.shaft(value);
   return driver.shaft();
}
```

cranefist, 2022-10-04:
Yes, that seems to be perfectly elegant solution. I was confused about the boolean uint8_t.. but apparently they can mix.

Thanks for the great work.. this seems to be now working perfectly. Also helps me to understand why you cant control this driver completely via UART. But this is the best compromise, only two wires just like without UART.

cranefist, 2022-10-04:
One thing i actually just noticed, the pulse counter stopped working.  This is still on the 0.28.0.

cranefist, 2022-10-04:
There still seems to be times when it does not change the direction, it also now happens always the same way. Always in the same direction.  When it should move forward, it moves backward. Never the other way around.

This did not happen with my brute force method. Maybe it was because i had that one click delay.. ill do some more testing.

I also switched to the latest master that included your last changes, just to see if it affected anything. No change.

gin66, 2022-10-04:
what is this driver ? any library ?

gin66, 2022-10-04:
After having a loop at the implementation of that driver, I recommend to move the actual bit toggling into another task and not in scope of FastAccelStepper’s callback. This would mean, that the callback e.g. sends a task signal or writes a variable, which the other task will use as input for toggling the bit. As soon as this toggle process is successful, a volatile variable can be set, which the callback will read on the next loop of FastAccelStepper.

cranefist, 2022-10-05:
I was testing something like that, i will experiment and try to find the best solution.

Thanks a lot for the help!

Your library works now, expect the pulse counter stopped working for some reason when using this direction function. But for me this issue is resolved!

gin66, 2022-10-05:
The attachable pulse counters are configured to read back the stepper pin and the direction pin. But this works only for hardware pins. As this is an add on, I have not added a check to prevent external direction pins to be used with pulse counter. eventually I can add this to the documentation.

closed 2022-10-05 by gin66

JanBluePix, 2023-06-10:
With WRITE_PERI_REG(GPIO_FUNC32_OUT_SEL_CFG_REG, READ_PERI_REG(GPIO_FUNC32_OUT_SEL_CFG_REG) | GPIO_FUNC32_OUT_INV_SEL);
I can basically reverse the direction of rotation of the stepper with the ESP32 (depending on which lens it should rotate). Also with TMCStepper.
A simple ->StepperXYsetInversRotation would be easier.

commit 81087103 (2022-10-02): try to fix #142 for esp32 mcpwm/pcnt and format code