gin66/FastAccelStepper #146: MKS TMC 2160-OC Stealth mode abnormal symptoms...

MuKwang, 2022-10-22:
I'm using it by connecting to MKS TMC 2160-OC

There is no problem in normal mode, but the noise of the motor increases a lot when used in stealth mode.

I'd like to know if it's a problem with MKS TMC 2160-OC or how it can be done in the FastAccelStepper library.

gin66, 2022-10-22:
That IC seems to be quite configurable. From the data sheet it says: `DIR Interface The motor is controlled by a step and direction input. Active edges on the STEP input can be rising edges or both rising and falling edgesas controlled by another mode bit (dedge).`

So make sure, that STEP is performed only on low to high transition of the step pin. This edge is controlled well by the FastAccelStepper, while the high to low transition is not. The latter may cause noise. In addition please choose a high micro stepping value.

Anyway, I have no experience with that driver, so this is just a bold guess.

MuKwang, 2022-10-23:
Well, 64 is the best for microsteps, but it doesn't seem to change much even if you raise it.

gin66, 2022-10-23:
Good. Unfortunately I cannot help you here. In case you find a solution or root cause, please post it here,

For now can close this issue.

closed 2022-10-23 by gin66