gin66/FastAccelStepper #150: speed is not updated correctly when stepper is already moving

DreiDe, 2022-11-11:
I'm changing the speed of an ESP32 controlled moving stepper multiple times a second in intervals between ~100Us/Pulse - 40000Us/Pulse without a real acceleration phase in between. While at the beginning the stepper keeps following the set speed as expected, after some changes further updates are ignored and the old speed is kept.

From a feeling I would say it takes maximum 10 updates until it starts happening and higher rotation speeds are more likely to persist, however I couldn't find any real pattern. When the stepper movement is stopped and restarted speed updates again work at first but are ignored then.

This behavior occurs no matter if I'm using `runForward()`, `runBackward()`, `moveTo()`, `move()` `setSpeedInUs()`, `setSpeedInMilliHz`.

Did you ever experience this behavior @gin66? Or do you have any idea what the cause could be? And especially how it can be prevented/fixed?

```cpp
#include "FastAccelStepper.h"

// As in StepperDemo for Motor 1 on ESP32
#define dirPinStepper 14
#define stepPinStepper 17

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper;

void setup() {
    Serial.begin(115200);
    engine.init();

    // pins are set to outputs here automatically
    stepper = engine.stepperConnectToPin(stepPinStepper);
    stepper->setDirectionPin(dirPinStepper);
    stepper->setAcceleration(1000000);
}

void loop() {
    int targetSpeed = rand() % 40000 + 100;
    stepper->setSpeedInUs(targetSpeed);  // the parameter is us/step !!!
    stepper->runForward();
    delay(500);
    Serial.printf("target: %d\n", targetSpeed);
    Serial.printf("actual: %d\n", stepper->getCurrentSpeedInUs());
}
```

```
target: 6033
actual: 6033
target: 17843
actual: 17843
target: 16362
actual: 16362
target: 31629
actual: 31629
target: 29800
actual: 29800
target: 35608
actual: 35608
target: 29852
actual: 29852
target: 7356
actual: 7356
target: 17356
actual: 17356
target: 38219
actual: 17356
target: 19811
actual: 17356
target: 28451
actual: 17356
target: 12943
actual: 12943
target: 38805
actual: 38805
target: 34208
actual: 34208
target: 26493
actual: 26493
target: 38430
actual: 38430
target: 17466
actual: 17466
target: 12269
actual: 12269
target: 38032
actual: 38032
target: 27017
actual: 27017
target: 29947
actual: 29947
target: 18072
actual: 18072
target: 32968
actual: 32968
target: 7080
actual: 7080
target: 26323
actual: 26323
target: 38649
actual: 26323
target: 7692
actual: 7692
target: 8264
actual: 8264
target: 16269
actual: 8264
target: 6651
actual: 6651
target: 30527
actual: 30527
target: 36290
actual: 30527
target: 22724
actual: 22724
target: 735
actual: 735
target: 5020
actual: 735
target: 28044
actual: 735
target: 39410
actual: 735
target: 29962
actual: 735
target: 38584
actual: 735
target: 17463
actual: 735
target: 3401
actual: 735
target: 34810
actual: 735
target: 27336
actual: 735
target: 31976
actual: 735
target: 30531
actual: 735
target: 20029
actual: 735
target: 497
actual: 497
target: 20775
actual: 706
target: 10591
actual: 706
target: 24290
actual: 706
target: 19444
actual: 706
target: 18234
actual: 706
target: 28525
actual: 706
target: 1729
actual: 706
target: 4130
actual: 706
target: 12827
actual: 706
target: 29226
actual: 706
target: 21843
actual: 706
target: 7434
actual: 706
target: 14204
actual: 706
target: 34860
actual: 706
target: 29849
actual: 706
target: 5720
actual: 706
target: 30356
actual: 706
target: 31032
actual: 706
target: 35672
actual: 706
target: 11713
actual: 706
target: 2590
actual: 706
target: 34609
actual: 706
target: 7219
actual: 706
target: 17505
actual: 706
target: 35795
actual: 706
target: 14149
actual: 706
target: 30427
actual: 706
target: 16819
actual: 706
target: 33597
actual: 706
target: 6924
actual: 706
target: 12696
actual: 706
target: 18749
actual: 706
target: 7456
actual: 706
target: 14284
actual: 706
target: 23193
actual: 706
target: 22345
actual: 706
target: 21107
actual: 706
target: 19406
actual: 706
target: 26609
actual: 706
target: 2854
actual: 706
target: 20452
actual: 706
target: 25765
actual: 706
target: 16883
actual: 706
target: 38838
actual: 706
target: 10901
actual: 706
target: 19790
actual: 706
target: 1430
actual: 706
target: 7437
actual: 706
target: 28295
actual: 706
target: 36756
actual: 706
target: 16063
actual: 706
target: 2111
actual: 706
target: 19527
actual: 706
target: 142
actual: 142
target: 7206
actual: 706
target: 1068
actual: 706
target: 6312
actual: 706
target: 39101
actual: 706
target: 34610
actual: 706
target: 2580
actual: 706
target: 4758
actual: 706
target: 6671
actual: 706
target: 23431
actual: 706
target: 914
actual: 706
target: 24947
actual: 706
target: 10664
actual: 706
target: 13297
actual: 706
target: 23725
actual: 706
target: 28538
actual: 706
target: 21031
actual: 706
target: 38118
actual: 706
target: 38587
actual: 706
target: 10251
actual: 706
target: 9287
actual: 706
target: 10013
actual: 706
target: 4279
actual: 706
target: 27095
actual: 706
target: 31850
actual: 706
target: 11850
actual: 706
target: 5013
actual: 706
target: 15662
actual: 706
target: 11234
actual: 706
target: 2373
actual: 706
target: 3647
actual: 706
target: 18621
actual: 706
target: 24930
actual: 706
target: 39240
actual: 706
target: 5657
actual: 706
target: 17778
actual: 706
target: 14826
actual: 706
target: 18603
actual: 706
target: 30697
actual: 706
target: 508
actual: 508
target: 4993
actual: 706
target: 14088
actual: 706
target: 19338
actual: 706
target: 37185
actual: 706
target: 193
actual: 193
target: 37288
actual: 706
target: 22820
actual: 706
target: 3311
actual: 706
target: 38846
actual: 706
target: 12487
actual: 706
target: 15810
actual: 706
target: 39309
actual: 706
target: 35987
actual: 706
target: 18768
actual: 706
target: 23203
actual: 706
target: 3573
actual: 706
target: 32000
actual: 706
target: 7774
actual: 706
target: 15205
actual: 706
target: 36283
actual: 706
target: 21052
actual: 706
target: 24470
actual: 706
target: 31887
actual: 706
target: 39402
actual: 706
target: 7510
actual: 706
target: 22005
actual: 706
target: 32210
actual: 706
target: 25500
actual: 706
target: 38096
actual: 706
target: 30242
actual: 706
target: 18685
actual: 706
target: 31960
actual: 706
target: 37147
actual: 706
target: 24024
actual: 706
target: 5831
actual: 706
target: 1258
actual: 706
target: 35486
actual: 706
target: 15319
actual: 706
target: 36500
actual: 706
target: 3515
actual: 706
target: 4155
actual: 706
target: 1782
actual: 706
target: 9974
```

gin66, 2022-11-12:
Thanks for reporting and same issue here. I have added it as another example. Just with my hw configuration.

Still investigating about the rott cause. 
v0.24.2 works
v0.25.4 works
v0.25.5 works
v0.25.6 fails
v0.25.7 fails

gin66, 2022-11-12:
v0.28.2 will fix this issue.

For a given acceleration, the speed jumps from 0 to a value appropriate for the first step (call it v1). If now the requested new speed v is lower than this first step speed v1, then this speed setting was just ignored. The fix was to set the speed to the requested speed v, if the ramp generator cannot reduce the speed anymore.

DreiDe, 2022-11-12:
Ok, great, that was an instant fix :D 
Thanks a lot for your quick support! I just tested it and can confirm that the speed update works again.

closed 2022-11-12 by DreiDe