gin66/FastAccelStepper #151: Can not reach 76 kSteps/s with arduino nano. Speeds does not seem to increase at higher speeds than 250 us /s

EyemationRens, 2022-12-06:
Hi, 

I'm not sure if this really is an issue or just me not understanding the library fully. In the documentation I read up to 76 kSteps/s or 37 kSteps/s with 2 steppers. When i lower the speed setting from 500 to 250 us/s i can see the speed increasing but when i go lower than 250 the speeds does not increase anymore. 

Is there a reason i can not go any faster ? i am using an Arduino nano

ps really like what you are doing here. Keep up the good work!

EyemationRens, 2022-12-07:
I had the acceleration not high enough,

closed 2022-12-07 by EyemationRens

gin66, 2022-12-07:
Good to hear and thanks for letting me know. Based on your issue, I have checked for any automated test case in regard to higher speeds with AVR. As I could not find one and one test case was erroneously anyway, I have changed that one to check the 37kSteps/s with two steppers.

gin66, 2022-12-07:
I have added another test case for 75kSteps/s (cannot reach 76000Hz).
And another test case, because I have stumbled about the issue, that with high acceleration, steps are lost.
For that issue #152 has been added

EyemationRens, 2022-12-08:
it was not really an issue, but i did not calculate how long it would take to reach the max speed. The motor had to start ramping down before it finished accelerating. Thus never reaching the max speed

Perhaps the .moveTo() could return a number to indicatie it will never reach the maximum speed with the acceleration and position setpoint settings

gin66, 2022-12-08:
Unfortunately getting this info out of moveTo is more complex due to the structure of the driver. While running you could monitor rampState(), if it turns to COAST anytime.