gin66/FastAccelStepper #166: Not working at wokwi.com

lesalbum, 2023-04-10:
Hello gin66,
First of all, thank you for the library itself and the support 👍.
I have not been able to compile successfully at wokwi.com. I would highly appreciate if someone can have a look at the errors that I am facing below:
My project link: https://wokwi.com/projects/361607351418946561

gin66, 2023-04-10:
Interesting website. From what I see, the simulator does not emulate the essential hardware blocks (mcpwm, pcnt or rmt). So it cannot be expected, that FastAccelStepper is working within that simulated environment.

lesalbum, 2023-04-11:
> Interesting website. From what I see, the simulator does not emulate the essential hardware blocks (mcpwm, pcnt or rmt). So it cannot be expected, that FastAccelStepper is working within that simulated environment.

I changed esp32 by Arduino and the library worked: https://wokwi.com/projects/361670237243500545
I think esp32 does not fully emulate its real function. btw, thank you so much for reply.

gin66, 2023-04-11:
yes, there is a full [emulator](https://github.com/buserror/simavr) available for the avr chips, which is used in the test suite of FastAccelStepper. Not sure, which kind of emulator wowki.com is using for avr, but it can be expected to be way more complete as their esp32 emulator.

closed 2023-04-11 by gin66