gin66/FastAccelStepper #169: not working (non-blocking) with esp32

lesalbum, 2023-05-03:
Hello gin66,
First of all, thank you for the library itself and the support 👍.
I can run stepper motor in blocking (moveTo(1000, true)) but I have done my best but have not been able to run in non-blocking.
I also replaced other ESP32 or stepper drivers (a4988, tmc2208).
I would highly appreciate if someone can have a look at the errors.

gin66, 2023-05-03:
looking where ?

lesalbum, 2023-05-04:
> looking where ?

the stepper motor ran slowly (about 1 round per 2 second, despite setSpeedInHz(600)) and the temperature of VMOT pin was very hot.

gin66, 2023-05-04:
if you use microstepping of 16 with a standard 200step/revolution stepper, then this means, that you have 3200steps/revolution. at 600Hz you can expect one revolution per 5.17s.

lesalbum, 2023-05-05:
> if you use microstepping of 16 with a standard 200step/revolution stepper, then this means, that you have 3200steps/revolution. at 600Hz you can expect one revolution per 5.17s.

I used the same set up, change the code only to run in blocking or non-blocking way. Pin MS1, MS2 and MS3 of A4988 driver are LOW.

gin66, 2023-05-05:
I would not recommend full step for A4988.

Anyway, you would need to share your code in order to identify the problem.

lesalbum, 2023-05-06:
> I would not recommend full step for A4988.
> 
> Anyway, you would need to share your code in order to identify the problem.

I set MS2 pin at HIGH means microstepper is 4. 
Here is the code in blocking way, [video ](https://drive.google.com/file/d/1IyawD5jBgA5fLNRLG8AegGqrNp0CdlAN/view?usp=sharing):
`#include "FastAccelStepper.h"
// As in StepperDemo for Motor 1 on ESP32
#define dirPinStepper 21
#define stepPinStepper 19

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  stepper->setDirectionPin(dirPinStepper);
  stepper->setAutoEnable(true);

  stepper->setSpeedInHz(1600);
  stepper->setAcceleration(800);
}

void loop() {
  stepper->moveTo(1600, true);
  stepper->moveTo(0, true);
}`

Here is the code in non-blocking way, [video](https://drive.google.com/file/d/1VHThAVNig6MKOSZGqCYJH3ZpHQz0KLUH/view?usp=sharing)
`#include "FastAccelStepper.h"
// As in StepperDemo for Motor 1 on ESP32
#define dirPinStepper 21
#define stepPinStepper 19

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  engine.init();
  stepper = engine.stepperConnectToPin(stepPinStepper);
  stepper->setDirectionPin(dirPinStepper);
  stepper->setAutoEnable(true);

  stepper->setSpeedInHz(1600);
  stepper->setAcceleration(800);
}

void loop() {
  stepper->moveTo(1600, false);
  stepper->moveTo(0, false);
}`

gin66, 2023-05-06:
Thanks for sharing the code. In non-blocking mode, the program continues to execute after the moveTo. The non-blocking code asks the stepper to move to 1600 and one moment later to return to 0. The stepper is interrupting the two moveTo randomly ever 4ms. So the stepper will be requested randomly to move to 0 or 1600. What would be the expected outcome ?

In blocking  mode, the moveTo ensures, that the move is executed. In non-blocking mode, this has to be ensured by the application. Check for reference (https://github.com/gin66/FastAccelStepper/issues/90)

lesalbum, 2023-05-07:
> Thanks for sharing the code. In non-blocking mode, the program continues to execute after the moveTo. The non-blocking code asks the stepper to move to 1600 and one moment later to return to 0. The [...]

Excellent explanation, thank you so much for your support.
Have a good day.

closed 2023-05-07 by lesalbum