gin66/FastAccelStepper #170: building problem

Elpablito, 2023-05-16:
HI
Building UsageExample I had thr following problem
C:\Users\gsrad\Documents\Arduino Sketch\libraries\FastAccelStepper\src\FastAccelStepper.cpp:627:22: error: 'max' was not declared in this scope
   _off_delay_count = max(delay_count, (uint16_t)1);
                      ^~~
C:\Users\gsrad\Documents\Arduino Sketch\libraries\FastAccelStepper\src\FastAccelStepper.cpp:627:22: note: suggested alternative: 'fmax'
   _off_delay_count = max(delay_count, (uint16_t)1);
                      ^~~
                      fmax

Is it true or a my problem
Thank

gin66, 2023-05-16:
Thanks for the hint. There is no issue with platformio - even so, the github actions to check correct build of all demos was broken. So I have fixed the build-system and in addition consequently replaced all `max()` by `fas_max()`. On platformio it still builds as before, but perhaps on arduino IDE this is a fix.

This is incorporated in the 0.29.2 (not released yet). As soon as the arduino library manager team has fixed/identified the issue, that latest releases are not showing up, the 0.29.2 will be published. Until then, please build directly from the github-project.

Elpablito, 2023-05-17:
Hi
Thank for your fast answer I have changed max() with fas_max(),it is OK, but now the compiler does not like ' #error '. 
I use sloeber-arduino-eclipse-plugin

C:\Users\gsrad\Documents\Arduino Sketch\libraries\FastAccelStepper\src\StepperISR_esp32.cpp:65:2: error: #error "Nothing defined here"
 #error "Nothing defined here"
  ^~~~~

gin66, 2023-05-17:
which esp32 derivate do you use ? apparently none of the defines recognize a valid esp32 derivate. Perhaps that toolchain defines a preprocessor variable named TEST ?

Elpablito, 2023-05-18:
Hi 
Sorry i don't know where to look for an anwer, If it is of some help i include the complite log.
[log fsatAcc.txt](https://github.com/gin66/FastAccelStepper/files/11508619/log.fsatAcc.txt)

gin66, 2023-05-18:
Thanks. Looks like this define from the compilation is wrong:
`-DARDUINO_ARCH_ARDUINO-ESP32`

The correct way to define the arduino architecture is:
`-DARDUINO_ARCH_ESP32`

How/where to define this in your build system, I cannot tell you.

Apparently the IDF-way of building FastAccelStepper still does not work, but as long as no one can provide a platformio-build template application, it will stay like this.

Elpablito, 2023-05-20:
Hi
Thank for help. I shall work

closed 2023-05-20 by gin66