gin66/FastAccelStepper #173: Error using move after forced stop

jie326513988, 2023-05-27:
Thank you for providing an excellent stepper motor library！

esp32-wroom-32D
Use stepper ->move (-100000),
Then during the runtime, I used forceStop() once from the button interrupt,
The motor stopped, but,
When I use stepper ->move (100), the motor moves, but
It is still executing the step per>move (-100000) operation.

if (get_key(KEY_BACK) == 1) //Press the button to return to the origin and execute it only once
{
    pinMode(SIG1, INPUT_PULLUP);
    delay(10);
    attachInterrupt(SIG1, return_to_origin, FALLING);
    stepper->move(-100000);
    // stepper->runForward();
}
void return_to_origin()
{
    if (digitalRead(SIG1) == 0)
    {
        //stepper->stopMove();
        //stepper->runForward();
        //stepper->forceStop();
        //stepper->setCurrentPosition(0);
        stepper->forceStopAndNewPosition(0); 
        detachInterrupt(SIG1); 
    }
}

if (get_key(KEY_UP) == 1) //Forward button
{
    stepper->move(100); 
}

jie326513988, 2023-05-27:
I think I found the problem. After forceStop(), the return value of targetPos() is not 0. Can you tell me how to reset it to zero

gin66, 2023-05-27:
Thanks for reporting. Hopefully the problem is now fixed with pre-0.29.3 aka master branch

gin66, 2023-05-27:
BTW:
- after `forceStop()` the position should be kept.
- after `forceStopAndNewPosition(pos)` the position should be `pos`

jie326513988, 2023-05-29:
Thank you for your help!
I have updated the latest library.
ForceStop() and forceStopAndNewPosition (0) still cannot execute move() correctly.
I have printed some information on loop(), hoping it will be helpful to you.

Void return_to_origin_loop()
{
    //return_to_origin() is triggered, returnToOrigin_loop will be set to 1
    If (returnToOrigin_loop==0) return;

    Serial. print ("targetPos:"); Serial. println (stepper ->targetPos());
    Serial. print ("getPositionAfterCommandsCompleted:"); Serial. println (step ->getPositionAfterCommandsCompleted());

    Stepper ->setPositionAfterCommandsCompleted (0);

    Serial. print ("getPositionAfterCommandsCompleted:"); Serial. println (step ->getPositionAfterCommandsCompleted());
    Serial. print ("targetPos:"); Serial. println (stepper ->targetPos());

    If (stepper ->isRunning()==0) returnToOrigin_ Loop=0;
}

Output information:
TargetPos: -100000
GetPositionAfterCommandsCompleted: -11283
GetPositionAfterCommandsCompleted: 0
TargetPos: -88717

After using runBackward() to return to the origin, move() is normal.

jie326513988, 2023-05-29:
When I first started up and used runBackward() to return to the original point,
After executing forceStop() and setCurrentPosition (0), move() can be used normally.
But executing forceStop() and setCurrentPosition (0) after move() will result in a move() error.

gin66, 2023-05-29:
Thanks for reporting. I have created a failing test case based on your report with setCurrentPosition(). I am now analyzing for the root cause.

gin66, 2023-05-29:
Please try the lastest version on master. At least it passes the test case.

jie326513988, 2023-05-31:
That's really good. The problem has been resolved.
thanks

closed 2023-05-31 by gin66