gin66/FastAccelStepper #174: Missing steps ESP32 S2 in high dynamic usage

ChrGri, 2023-05-27:
Hello and thank you for this awesome library.

This library is used with an ESP32 S2 Mini in my DIY active pedal (https://github.com/ChrGri/DIY-Sim-Racing-Active-Pedal/blob/main/Arduino/Esp32_S2/Main/Main.ino). 
Here, a Loadcell measures force applied to the pedal and uses a closed loop stepper (https://www.omc-stepperonline.com/de/nema-23-integrierter-easy-servo-motor-180w-3000rpm-0-6nm-84-98oz-in-20-50vdc-buerstenloser-dc-servomotor-isv57t-180) to adjust the pedal position. Unfortunately, after some usage, the pedal seems to lose its neutral position, thus the pedal tilts more and more to the ground and I'm trying to understand the root cause of this problem. 

The cycle time of the code is approx. 130 us and depending on the loadcells reading, a new non-blocking moveTo(..., False) is executed. The position and direction might significantly differ from the cycle before. Stepper speed and acceleration are configured rather high as required by the application.

Step+ and Dir+ of the motor are wired to 3V3 and Step- and Dir- are wired to the GPIOs. I wonder, if applied in this configuration, some stepper code logic needs to be changed from rising to falling edge?
Other remarks of potential solutions are highly welcome.

BR

gin66, 2023-05-27:
If I understand the datasheet correctly, then the DIR- and PUL- should be connected to GND.

In order to support the debugging, I would advise:
1. which position does FastAccelStepper report ? 
2. if you remove the load, do you still observe the deviation after some time ?

ChrGri, 2023-05-27:
> If I understand the datasheet correctly, then the DIR- and PUL- should be connected to GND.
> 
> In order to support the debugging, I would advise:
> 
> 1. which position does FastAccelStepper report ?
> 2. if you remove the load, do you still observe the deviation after some time ?

Thank you @gin66 for the fast reply. I've tried grounding the DIR- and Step- pins first, but then the ESP32 was not starting. Therefore I used the GPIO as ground. 

Calling "stepper->getCurrentPosition();" gives the drifted position (e.g. 0 although the pedal clearly is out of the initial position) unfortunately. I will try without load next. 

BR

ChrGri, 2023-05-28:
@gin66 I've removed the load from the stepper. When I ask for high velocity & acceleration movements with direction changes in between the cycles, sometimes the stepper moves in the wrong direction. 

I've build a new µC where the Dir- and Step- are connected to GND and Dir+ and Step+ connected to the GPIOs as suggested --> no difference.

ChrGri, 2023-05-28:
I've read a article from stepperonline which mentions that they have tested 3V3 for their closed loop system (https://www.omc-stepperonline.com/support/do-your-stepper-drivers-accept-33v-signal-voltage-for-arduino-and-raspberry-pi). 

I do have a logic analyzer, which I'm using next.

I don't need acceleration. How do I disable it in the library?

gin66, 2023-05-28:
do you have a short working code, which reveals the issue, so I can test it here ? Without all the peripherals in the code, which I do not have. This means, having only the stepper code would be of great help to see, if there is any problem with the lib.

Depending on the logic analyzer, perhaps the load of the driver is just too high and even the logic analyzer detects sometimes low. Anyway, the manufacturer does not recommend 3.3V operation. Already some ground shifts or long wires may impact the outcome.

I have thought for some time to disable acceleration altogether e.g. by setting acceleration to 0, but haven’t implemented it.

ChrGri, 2023-05-28:
I've tested without the stepper driver cables attached and observed similar behaviour. 

I will prepare some test code without the other dependencies now.

gin66, 2023-05-28:
Thanks for the measurement and code example. This is very helpful

Quite helpful. I have modified the code with some comments: (https://github.com/gin66/FastAccelStepper/blob/20a8301b4b8c4ee2bbb4e657049f4c0df26045b7/examples/Issue174/Issue174.ino)

The issue is, that you run the loop with cycle time of couple of 100 us. The stepper gets every cycle a new position. Most of the positions, the stepper task does not even see. The stepper task looks every 4ms for commands and all former position requests are overwritten. That's why, I have added print out of the `stepper->targetPos()`. This way you can see, what the stepper is intending to run to.

The max speed in your system is 13333 steps/s. So at full speed at 130us loop cycle time, the stepper can run only 1 or 2 steps. But every loop requests to run 500 steps during that time. This does not work. 
Consequently: the stepper needs time to run.

ChrGri, 2023-05-28:
Thanks for the testcase. Hardware wise, the stepper can handle up to 4k RPM and 300kHz step rates. 

Given that the steppers trajectory is updated every 4ms and the sawtooth from the test has a period of 2.6ms = 130us/cycle * 10000 steps/tooth /500steps/cycle, I would have expected the stepper to be mostly stationary and move forward and backwards only slightly. But since the stepper kept rotating in one direction, could it be, that the ramp generator planner loses track of its own states somewhere?

Real world usage should not be as demanding as this testcase. However, high frame rates are required, otherwise the pedal will start to oscillate.

gin66, 2023-05-28:
There is still the possibility, that on esp32 something goes wrong or something is not ok with the interface to that motor, which causes position divergence. Today I have made the test using simavr, which emulates an AtMega328p, and it has run well. Testing on esp32 takes more time as I need to prepare and attach the esp32 hardware.

On ESP32 the rmt module is used, which I believe to have better performance than mcpwm/pcnt, but rmt is most likely less in use in the field. In regard to steps lost: this should not happen at all, if neither ForceStopAndNewPosition nor setCurrentPosition has been called. But I still have doubts about the motor interface  3,3V/5V, so I would rather start with this.

BTW: cool project of yours and nice video.

gin66, 2023-05-28:
Pin losses could be caused by DIR/STEP timing. FastAccelStepper supports to add a delay on direction change:
```
setDirectionPin(dirPinStepper, false , MIN_DIR_DELAY_US);
```
Perhaps this helps

ChrGri, 2023-05-29:
Thank you very much. I will try the test on an Arduino Nano later that day. 

Do you have an idea which delay value on direction changes I should try with the ESP32?

Thank you for this library and if you need an ESP32 S2 Mini (15€/3 units) for testing, I'd be happy to help e.g. via Amazon wishlist or PayPal.

gin66, 2023-05-29:
In your example code, the acceleration is 100. This means direction changes will not happen fast and your assumption "Channel 4 should follow channel 3 somehow" is not valid.

I have updated the example in the Issue174-branch to match it to your code. 

I think the scaling in the image is not correct, but at least it is clearly visible, that DIR A is changing:
[image]

gin66, 2023-05-29:
The code, which I have used is [here](https://github.com/gin66/FastAccelStepper/blob/ed1fed491edeb3d900b0238c105f3ae85f3a6cc8/examples/Issue174/Issue174.ino). 

Could you please check and tell me, which values I should change ?

ChrGri, 2023-05-29:
@gin66 I'm so sorry. I've to withdraw my statement, that it's not working on the Atmel as well. Uploaded the same C code to Arduino Nano and ESP32 S2 mini and let them run in parallel. The output of the logic analyzer is different for the Arduino and indeed shows Dir changes. For the ESP32 this is not the case. 

[image]

See Ch0 vs Ch5.

**Arduino Nano:**
Ch 0: DIR+
Ch 1: Debug
Ch 2: STEP+

**ESP32 S2 mini:**
Ch 5: DIR+
Ch 6: Debug
Ch 7: STEP+

gin66, 2023-05-29:
OK. So I need to test directly on HW and the atmel simulator is not of help anymore.

Could you please do me a favor and check within the loop for `stepper->rampState()` ? I like to check, if the issue is related to the ramp generator or the low level driver. That ramp state value may of help.

ChrGri, 2023-05-29:
Sure. I've added the requested command like seen below:
[image]

Here are the two log files:
[AruinoNano.log](https://github.com/gin66/FastAccelStepper/files/11592628/AruinoNano.log)
[Esp32_S2_mini.log](https://github.com/gin66/FastAccelStepper/files/11592630/Esp32_S2_mini.log)

Not sure whether this is important or not, but the Dir pulses differe in length a lot. Ch 2 is Arduino and Ch 7 is the ESP:
[image]

I've setup the cycle time to be 500us for both µC in that example.

ChrGri, 2023-05-29:
And just for completeness, here is the logic analyzer output for cycle time of 500ms and acceleration=1e8 steps/s^2

[image]

The direction change can bee seen for ESP (CH 5) and Arduino (CH 0).

gin66, 2023-05-29:
Thanks for quick action.

The ramp states are defined like this:
```
// The defined ramp states are:
#define RAMP_STATE_IDLE 0
#define RAMP_STATE_COAST 1
#define RAMP_STATE_ACCELERATE 2
#define RAMP_STATE_DECELERATE_TO_STOP 4
#define RAMP_STATE_DECELERATE (4 + 8)
#define RAMP_STATE_REVERSE (4 + 16)
#define RAMP_STATE_ACCELERATING_FLAG 2
#define RAMP_STATE_DECELERATING_FLAG 4

// And the two directions of a move
#define RAMP_DIRECTION_COUNT_UP 32
#define RAMP_DIRECTION_COUNT_DOWN 64
```

So there are the following states in the esp32 log:
```
34 count down+accelerating
44 count down+decelerating
52 count down+reverse
66 count up+accelerating
76 count up+decelerating
84 count up+reverse
```
This means:
1. The stepper should go in both directions
2. At now point, the stepper reaches max. speed

And from your latest info, I understand, the stepper is changing direction, too.

So I currently cannot find any issue.

Could you please change the code, that the ramp is not executed so superfast, which gives FastAccelStepper a position to follow ?  Please try instead this code, which provides the stepper with the positions to run the ramp 10 times per second:
```
PositionNext = (currentTime / 10) % 10000;
```

ChrGri, 2023-05-29:
> And from your latest info, I understand, the stepper is changing direction, too.

Only for the example, where the cycle time was increased to 500ms and the acceleration increased to 1e8.
The post before, with the attached log files was created with cycle times of 500us.

gin66, 2023-05-29:
OK. Now I understand. Thanks for clarification.

Hope later today I can check on the esp32 (without S2)

ChrGri, 2023-05-29:
Good point, I've run the same test on the esp32 (without S2) and it's running great. Direction changes as expected.

[image]

So only the esp32 S2 seems to have the issue currently.

gin66, 2023-05-29:
The S2 uses RMT module and the stock esp32 a mcpwm/pcnt combo. But there is a possibility to force the use of the rmt module even for stock:
```
#define DRIVER_MCPWM_PCNT 0
#define DRIVER_RMT 1
#define DRIVER_DONT_CARE 2
  FastAccelStepper* stepperConnectToPin(uint8_t step_pin, uint8_t driver_type);
```
Just add the parameter `DRIVER_RMT` to the `stepperConnectToPin()`. On the esp32s2 will give a compile error.

ChrGri, 2023-05-29:
Good to know. I've flashed the esp32 (without S2) with either `DRIVER_MCPWM_PCNT` and `DRIVER_RMT`. CH 5 is Dir+, CH6 is Step+ and CH 7 is the debug signal.

`stepper = engine.stepperConnectToPin(stepPinStepper, DRIVER_MCPWM_PCNT);` gives this result:

[image]

`stepper = engine.stepperConnectToPin(stepPinStepper, DRIVER_RMT);`gives this result:

[image]

gin66, 2023-05-29:
With esp32-hw and your app, I could reproduce the issue with the rmt-module.

Funnily, there was even a comment in the rmt-module about the dir-change and stating an assumption. Apparently that assumption has not been met in your application. I have now applied a quick-fix, which allows the motor to change direction. I am just not sure, if this causes steps lost. That's why I have added pulse counter support for esp32s2, which can support the detection of lost pulses.

gin66, 2023-05-29:
the esp32 based hw tests indicate, that steps are lost due to that hotfix.

ChrGri, 2023-05-29:
You are absolutely awesome. I've flashed the update and it's already miles better than before. 

When flashing my testscript to my pedal, I indeed can confirm the lost steps. With every sawtooth the steppers 0 position moves constantly further from it's initial position. Wanted to count the forward and backwards steps via logic analyzer next.

gin66, 2023-05-30:
Thanks for the detailed investigation. The mcpwm/pcnt combo runs pretty well, while the rmt-version has some issues. In the meantime I have reworked the code and believe, that changes in direction is now better managed and step pauses as in your pictures are avoided. 

Most of my hw tests are passing. Just two similar test cases are constantly failing. 

So step loss has improved, so I have decided to upload the current code. Hope you can confirm an improvement, too.

ChrGri, 2023-05-30:
Thank you very much. I've flashed with the update library. It's much better, yet not perfect.

### RMT with **500ms** delay in between the bursts:
[image]
Observation: No steps lost

### RMT with **50ms** delay in between the bursts:
[image]
Observation: No steps lost

### RMT with **5ms** delay in between the bursts:
[image]
Observation: Steps are beeing lost, position is drifting.

### MCPWM with **5ms** delay in between the bursts:
[image]
Observation: Steps are beeing lost, position is **not** drifting.

For the tests, the stepper limits have been parameterized as follows:
stepper->setSpeedInHz(33333)
stepper->setAcceleration(1e8).

The required steprates to follow the burst with the given delay of [500ms, 50ms, 50ms]  has to be [1k, 10k, 100k]steps/s. Ignoring acceleration here.

I feel like, that RMT behaves perfect when the requested step rates are lower than the parameterized limit. But steps are lost when the requested steprate is higher than the parameterized limit. MCPWM, seems not to drift but oscilation around the center.

Does this observation relate to the failed testcases?

gin66, 2023-05-30:
Thanks for the measurements and it fits for the rmt. The two failing test cases are faster versions of a test, which is passing. So I need to focus on that.

For mcpwm/pcnt I have not observed step loss for a while and with my test cases. After rmt is ok, I will check that again. The issue with mcpwm/pcnt is, that high speed directly followed by a pause may lead to step loss under higher cpu load, because the interrupt has to stop the pwm right before the next step. At high speed of 100ksteps/s, this is <<10us. Using deceleration with not so high values, any possible overruns due to delayed interrupt service at high speed can be recovered, because the pulse counter detects this.

ChrGri, 2023-05-31:
@gin66 I'm using the ADC [lib](https://github.com/ADS1xxx-Series-ADC-Libraries/ADS1255-ADS1256/tree/56c52e1d884ace31d0188bcf04b5b9a225e74d71) to gather ADS1256 readings via SPI. Calling `stepper = engine.stepperConnectToPin(stepPinStepper);` corrupts the data reading or breaks the SPI communication to the ADC. Don't know exactly. The old implementation did not have that problem.

Apparently the call order has an influence.
It does **not** work, when I initialize the ADC first and the stepper afterwards.
It works, when I initialize the stepper first and the ADC afterwards.

It's not a blocking point for my project, since I only need to change the call order to make things work. But, perhaps, something you may want to keep an eye on.

gin66, 2023-05-31:
is it using Pin 18 ? I have forgotten to disable the probe for toggling this port pin in the rmt driver:
ˋˋˋ
#define TEST_PROBE_1 18
ˋˋˋ

ChrGri, 2023-05-31:
I've flashed the testcode to an esp32 & esp32 S2 for comparison. The results look different

### ESP32 with `engine.stepperConnectToPin(stepPinStepper, DRIVER_RMT)`
[image]
Observation: No steps lost

### ESP32 S2 mini:
[image]
Observation: Position is drifting.

gin66, 2023-05-31:
The new version passes all tests.

As I had once a test failure, perhaps there is still a sporadic issue, which needs to be monitored further. Please try the latest version

ChrGri, 2023-05-31:
Thank you. Flashed the the code to the pedal. Unfortunately, it seems to perform a lot worse than it did before. From observation I would say, that the direction signal isn't following the requested direction. 

I'm gonna try to write a testcase which is as close as possible to the application.

gin66, 2023-05-31:
Do you observe any step loss aka position drifting ?

ChrGri, 2023-05-31:
I think that it's fine if the requested step rate is lower than the parameterized maximum steprate. Once the requested steprate gets to large, it's much more drifting than before. I reperformed the sawtooth testpattern with [300, 5]ms delay per cycle.

### Esp32 S2 Delay 300ms
[image]
Observation: Position follows the sawtooth pattern

### Esp32 S2 Delay 5ms
[image]
Observation: Position drift away, significantly worse than before

gin66, 2023-05-31:
I have added your example with modification as Issue174.ino under examples. Here you can see, how to attach a pulse counter to check, if steps are lost.

The pulse generation runs flawless with acceleration 1e6.

With 1e9 there are immediately deviations and the esp32 stalls even after a short while, which is even more weird.
The log out of reset is:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:184
load:0x40078000,len:12680
load:0x40080400,len:2908
entry 0x400805c4
Setup stepper!
Return to 0 from 491
=> FAIL with pcnt=
2
Return to 0 from 439
=> FAIL with pcnt=
4
Return to 0 from 359
=> OK
Return to 0 from 243
=> FAIL with pcnt=
20
Return to 0 from 109
=> OK
Return to 0 from -37
=> FAIL with pcnt=
-5
Return to 0 from -170
=> FAIL with pcnt=
-1
Return to 0 from -305
=> OK
Return to 0 from -423
=> OK
Return to 0 from -486
=> OK
```

ChrGri, 2023-06-01:
That's weird indeed. Perhaps a numerical issue?
Are the failed testcases RMT exclusive?

gin66, 2023-06-01:
still have an issue in the rmt-driver. This will take some more time to fix

ChrGri, 2023-06-01:
> still have an issue in the rmt-driver. This will take some more time to fix

You already did an awesome job. Let me know, if you need any help.

ChrGri, 2023-06-04:
FYI: I've had significant amount of step loses on the pedal, when calling (a) for a move to the stepper->getCurrentPosition() or (b) same target position over and over again (150us cycle time). Don't know yet if (a) or (b) created that problem. Problems are gone with this command:
https://github.com/ChrGri/DIY-Sim-Racing-Active-Pedal/blob/33c2687d5ded583c01289e7f45100538fee369e3/Arduino/Esp32_S2/Main/Main.ino#L544

Haven't tried the command on the testcase yet.

gin66, 2023-06-04:
Good that you have found a solution.

Anyway, I have set up my esp32 with logic analyzer and modified your test case. Several bugs identified and fixed. Hope it applies to esp32s2, too.

So I have released 0.30.1 as I believe it to be much better. All esp32 hw based tests are passed and your test case (modified and stored as examples/Issue174/Issue174.ino) do not show any pulse loss anymore.

ChrGri, 2023-06-05:
Thanks a lot. I've tried back and forth, with the current mainline head, the stepper isn't moving anymore.

ChrGri, 2023-06-05:
I've only tried on my esp32 s2 today. I will try on a regular esp32 tomorrow.

gin66, 2023-06-05:
Please try to  apply this patch:

```
diff --git a/src/StepperISR_esp32_rmt.cpp b/src/StepperISR_esp32_rmt.cpp
index d459069..60899ba 100644
--- a/src/StepperISR_esp32_rmt.cpp
+++ b/src/StepperISR_esp32_rmt.cpp
@@ -492,6 +492,8 @@ void StepperQueue::startQueue_rmt() {

   // This starts the rmt module
   RMT.conf_ch[channel].conf1.tx_conti_mode = 1;
+
+  RMT.conf_ch[channel].conf1.tx_start = 1;
 }
 void StepperQueue::forceStop_rmt() {
   stop_rmt(true);
```

Perhaps the ESP32S2 needs the tx_start, which the esp32 does not need.

ChrGri, 2023-06-06:
Happy to confirm that patch is working :)

In a short test, the pedal felt great. I will spend more time this week trying to provoke step losses.

closed 2023-06-11 by gin66

ChrGri, 2023-06-19:
@gin66 Thx you for fixing the lost steps. Meanwhile, I've mounted a regular ESP32 on my pedal to be able to switch between RMT and MCPWM. The good news is, that none of them seem to have issues with step loss. However, the response of the stepper to dyanmic input seems to be quite different for RMT and MCPWM. In MCPWM the pedal response is perfect. No delay or latency whatsoever. In RMT, the response sometimes seems to be delayed and not smooth. Without knowing the root cause of the difference, I would describe the behaviour as, once the input queue is full, the target position is not updated by new commands until the queue is completed entirely.

gin66, 2023-06-19:
There is a difference in regard to direction changes. The RMT will do DIR change only, if the current command in the queue does not generate steps. If this is not the case, then the driver will add a pause, which may slow down in a highly dynamic use case.

Does this match your experience ?

ChrGri, 2023-06-19:
That's possible. Is there a technical reason for that functional difference?

gin66, 2023-06-20:
Yes, this arises from the current implementation. The RMT buffer is split into two parts and usually one half executes one command from the step queue. This means, while one half is being processed by the RMT HW, then the other buffer can be prepared with the next command. If now a command with direction change needs to be processed, the dir output needs to update. If the currently executing command is performing steps, it is not possible to safely change the dir output without risking one step in the wrong direction. So the driver requires the current running command to have no steps in order to change the direction pin. If this is not the case, then the driver inserts a pause.

Perhaps there are alternative ways to avoid this, just I have currently no good idea, how this can be done. Normally direction change appear at stopped motor, so I would expect most application to have no issue with this pause.

This raises the question, if this technical implementation causes the observed behavior or not.

ChrGri, 2023-06-20:
Thank you for the detailed explanation. Since I'm currently using the regular esp, this does not need prioritization tbh.

If you want to verify your assumption, I could trigger a serial print for specific conditions/state and try to correlate that with the pedal response. Do you have an idea which state to check (internal variable or function argument of the API)?

gin66, 2023-06-20:
direction changes could be a trigger. this you can get directly from the port pin