gin66/FastAccelStepper #176: V0.29.2 not setting positions

daniel-frenkel, 2023-06-03:
I recently upgraded from V0.28.4 to V0.29.2 and it no longer works.

I have not changed any other code, just the version.

When setting the move or moveTo function, the position always remains at 0.

Running the following code prints out a value of 0 and the motor does not begin to move.

```
stepper->moveTo(1000);
Serial.println(stepper->targetPos());
```

Downgrading to V28.4 works as expected.

gin66, 2023-06-03:
That's strange. Which µC are you using ?

daniel-frenkel, 2023-06-03:
ESP32

Here's the code:

```
    uint16_t set_pos = stepper->moveTo(10000);
    printf("set_pos: %i\n", set_pos);
```

Output:
```
set_pos: 0
```

gin66, 2023-06-03:
Good that 0.28.4 works for you.

I have downgraded to 0.28.4 and I cannot reproduce this problem. Can you please provide more details ?

gin66, 2023-06-04:
comment to this:
```
stepper->moveTo(1000);
Serial.println(stepper->targetPos());
```

The targetPos() is now updated by the stepper task, which may take up to 4ms to update the value returned by targetPos().

daniel-frenkel, 2023-06-07:
I switched over to Platformio and everything is working as expected so I'm not sure what was wrong. I'll close this

closed 2023-06-07 by daniel-frenkel