gin66/FastAccelStepper #177: separate parameter settings for acceleration and deceleration

JanBluePix, 2023-06-11:
Entering a new value for a acceleration as deceleration after the acceleration is done works as described in Issue 118. Setting a separate value for acceleration and deceleration would be nicer though.

gin66, 2023-06-11:
Agree, that this would be nicer for __some__ applications. Unfortunately based on current structure of the ramp generator, this is a quite complex endeavor. So there is no plan to start implementation in the coming weeks and more likely even months.

PRs are surely welcome.

gin66, 2025-08-10:
Still there are several places requiring a change and variables need to check for overruns….

gin66, 2025-08-10:
Strange. One earlier comment is lost from github…

To recap: Comparably easy would be an implementation to reduce the ramp steps for deceleration by e.g. 2, 4, 8,… This would be mean deceleration would be by that factor higher than acceleration. Just there are several places, where this needs to be changed and overruns/underruns need to be checked.

closed 2025-08-11 by gin66