gin66/FastAccelStepper #178: Reducing speed while running continously doesn't respect acceleration (ESP32)

genghisnico13, 2023-06-11:
I tested V0.30.3, set a speed, acceleration and runForward(), and if I increase speed either while accelerating or at desired speed everything works as expected,I can also change acceleration without problems, the possible Bug is when reducing the speed, if the new speed is lower than the current running speed when applying it, it seems to be set immediately(without respecting the acceleration).
I even tried with stopMove() and if the current running speed while decelerating is higher than the new speed it seems to apply it instantly.

Let me know if you need any more information or need me to test something.

Simple test code just in case:
` stepper->setSpeedInHz(3600);
  stepper->setAcceleration(320);
  stepper->runForward();
  delay(8000);
  stepper->setSpeedInHz(450);
  stepper->applySpeedAcceleration();`

gin66, 2023-06-12:
Thanks for the error report. The issue has been fixed in 0.30.4 and a new test case has been added.

genghisnico13, 2023-06-13:
Wow, thanks for the quick Fix, it works as expected. Thanks for making this library.

closed 2023-06-13 by genghisnico13