gin66/FastAccelStepper #180: Support for Raspberry pi

ajay01994, 2023-06-19:
Hi ,Thanks a lot for releasing this library , is it possible for you to support Raspberry Pi or could you tell how to port the existing code in RPI ?

gin66, 2023-06-19:
Do you mean the Pico ?

ajay01994, 2023-06-19:
Hi , No Raspberry pi 4B

gin66, 2023-06-19:
Thanks for clarification.

A port to raspberry pi will be quite difficult. The only HW support seems to be Hardware PWM on GPIO12, GPIO13, GPIO18, GPIO19, which may be similar to the AVR pwm solution of FastAccelStepper. This would require, that for every pulse an interrupt can be created and serviced in time. So no fancy pcnt or rmt as the esp32 offers. This means, only a interrupt intensive solution comes to my mind.

Just, which OS to support ? Linux or micropython or bare metal or whatever else is out there ? A device driver for linux would be a total different story than what FastAccelStepper is designed for. An implementation for micropython will again have not much similarity with FastAccelStepper.

raspberry pico is IMHO a different story and more similar to avr/esp32, 

So in short: I definitely will not port to raspberry pi, nor pico. Any PR will be appreciated for a Pico port.
Most reusable may be the ramp generator together with its test suite. The generator creates raw stepper commands (like generate 10 steps at a rate and change direction). What is necessary to be ported: 1) the low level driver to execute the commands and 2) means to cyclically call the ramp generator e.g. every 4ms. That’s all.

closed 2023-07-05 by gin66

ajay01994, 2023-07-09:
Thanks for your clarification ...