gin66/FastAccelStepper #184: rmt_memory_rw_rst

diarus, 2023-06-21:
Hello. Error of compilation. Help me fix this:
`.pio/libdeps/esp32dev/FastAccelStepper/src/StepperISR_esp32_rmt.cpp: In member function 'void StepperQueue::startQueue_rmt()':
.pio/libdeps/esp32dev/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:409:28: warning: 'esp_err_t rmt_memory_rw_rst(rmt_channel_t)' is deprecated: use rmt_tx_memory_reset or rmt_rx_memory_reset instead [-Wdeprecated-declarations]
   rmt_memory_rw_rst(channel);
                            ^
In file included from .pio/libdeps/esp32dev/FastAccelStepper/src/common.h:116,
                 from .pio/libdeps/esp32dev/FastAccelStepper/src/FastAccelStepper.h:5,
                 from .pio/libdeps/esp32dev/FastAccelStepper/src/StepperISR.h:3,
                 from .pio/libdeps/esp32dev/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:1:
C:/Users/User/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/rmt.h:898:11: note: declared here
 esp_err_t rmt_memory_rw_rst(rmt_channel_t channel)
           ^~~~~~~~~~~~~Archiving .pio\build\esp32dev\libda7\libFastAccelStepper.a
~~~~
.pio/libdeps/esp32dev/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:409:28: warning: 'esp_err_t rmt_memory_rw_rst(rmt_channel_t)' is deprecated: use rmt_tx_memory_reset or rmt_rx_memory_reset instead [-Wdeprecated-declarations]
   rmt_memory_rw_rst(channel);
                            ^
In file included from .pio/libdeps/esp32dev/FastAccelStepper/src/common.h:116,
Compiling .pio\build\esp32dev\FrameworkArduino\FirmwareMSC.cpp.o
                 from .pio/libdeps/esp32dev/FastAccelStepper/src/FastAccelStepper.h:5,
                 from .pio/libdeps/esp32dev/FastAccelStepper/src/StepperISR.h:3,
                 from .pio/libdeps/esp32dev/FastAccelStepper/src/StepperISR_esp32_rmt.cpp:1:
C:/Users/User/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/driver/include/driver/rmt.h:898:11: note: declared here
 esp_err_t rmt_memory_rw_rst(rmt_channel_t channel)
           ^~~~~~~~~~~~~~~Compiling .pio\build\esp32dev\FrameworkArduino\FunctionalInterrupt.cpp.o`

Problems: 'esp_err_t rmt_memory_rw_rst(rmt_channel_t)' is deprecated: use rmt_tx_memory_reset or rmt_rx_memory_reset instead [-Wdeprecated-declarations]/
**Platformio, Arduino framework, FastAccelStepper ver.0.30.4**
![2023-06-21_22-24-40](https://github.com/gin66/FastAccelStepper/assets/137332935/d831de5d-407a-4ee5-9283-f3ac295672b4)
![2023-06-21_22-23-40](https://github.com/gin66/FastAccelStepper/assets/137332935/ffe9159d-78c4-4a42-babf-b75bdd656de5)

diarus, 2023-06-21:
[image]

diarus, 2023-06-21:
[image]
[image]

gin66, 2023-06-21:
which espressif sdk version do you use?

diarus, 2023-06-22:
Espressif 32  6.3.1, Platformio, 
Boards: Espressif ESP32 Dev Module

Platformio.ini:
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	gin66/FastAccelStepper@^0.30.4
	gyverlibs/EncButton@^2.0

diarus, 2023-06-23:
I found the reason for the error: the error occurs if there is an output to the port in the code:
Serial.begin(9600) and Serial.println("click") .  The port contacts are free in my project.
On the arduino Leonardo board, this problem does not arise.

gin66, 2023-06-23:
Good, that you have found a solution. Even though, the Serial should be not related to that error.

Every build is tested by github actions using this definition for esp32:
```
[env:esp32]
platform    = espressif32
board       = esp32dev
framework   = arduino
build_flags = -Werror -Wall -Wno-deprecated-declarations -Wno-error=incompatible-pointer-types
board_build.f_cpu = 240000000L
lib_extra_dirs = ../../..
```

And platformio packages are:
```
Resolving esp32 dependencies...
Platform espressif32 @ 2023.4.2 (required: espressif32)
├── framework-arduinoespressif32 @ 2.0.8 (required: https://github.com/tasmota/arduino-esp32/releases/download/2.0.8/framework-arduinoespressif32.zip)
├── framework-espidf @ 3.40404.20230310 (required: https://github.com/tasmota/esp-idf/releases/download/v4.4.4.20230310/esp-idf-v4.4.4.zip)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
├── tool-esptoolpy @ 1.40501.0 (required: https://github.com/tasmota/esptool/releases/download/v4.5.1/esptool-4.5.1.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── toolchain-esp32ulp @ 1.23500.220830 (required: platformio/toolchain-esp32ulp @ ~1.23500.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5)

Resolving esp32arduinolatest dependencies...
Platform espressif32 @ 2023.4.2 (required: espressif32)
├── framework-arduinoespressif32 @ 2.0.8 (required: https://github.com/tasmota/arduino-esp32/releases/download/2.0.8/framework-arduinoespressif32.zip)
├── framework-espidf @ 3.40404.20230310 (required: https://github.com/tasmota/esp-idf/releases/download/v4.4.4.20230310/esp-idf-v4.4.4.zip)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
├── tool-esptoolpy @ 1.40501.0 (required: https://github.com/tasmota/esptool/releases/download/v4.5.1/esptool-4.5.1.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── toolchain-esp32ulp @ 1.23500.220830 (required: platformio/toolchain-esp32ulp @ ~1.23500.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5)

Resolving esp32arduinoV340 dependencies...
Platform espressif32 @ 3.4.0 (required: espressif32 @ 3.4.0)
├── framework-arduinoespressif32 @ 3.10006.210326 (required: platformio/framework-arduinoespressif32 @ ~3.10006.0)
├── tool-esptoolpy @ 1.30100.210531 (required: platformio/tool-esptoolpy @ ~1.30100.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
└── toolchain-xtensa32 @ 2.50200.97 (required: platformio/toolchain-xtensa32 @ ~2.50200.0)

Resolving esp32idf dependencies...
Platform espressif32 @ 2023.4.2 (required: espressif32)
├── framework-arduinoespressif32 @ 2.0.8 (required: https://github.com/tasmota/arduino-esp32/releases/download/2.0.8/framework-arduinoespressif32.zip)
├── framework-espidf @ 3.40404.20230310 (required: https://github.com/tasmota/esp-idf/releases/download/v4.4.4.20230310/esp-idf-v4.4.4.zip)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
├── tool-esptoolpy @ 1.40501.0 (required: https://github.com/tasmota/esptool/releases/download/v4.5.1/esptool-4.5.1.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
├── toolchain-esp32ulp @ 1.23500.220830 (required: platformio/toolchain-esp32ulp @ ~1.23500.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5)

Resolving esp32s2 dependencies...
Platform espressif32 @ 2023.4.2 (required: https://github.com/tasmota/platform-espressif32/releases/download/2023.04.02/platform-espressif32.zip)
├── framework-arduinoespressif32 @ 2.0.8 (required: https://github.com/tasmota/arduino-esp32/releases/download/2.0.8/framework-arduinoespressif32.zip)
├── framework-espidf @ 3.40404.20230310 (required: https://github.com/tasmota/esp-idf/releases/download/v4.4.4.20230310/esp-idf-v4.4.4.zip)
├── tool-cmake @ 3.21.3 (required: platformio/tool-cmake @ ~3.21.0)
[... 8 lines cut ...]

Resolving esp32s3 dependencies...
Platform espressif32 @ 2023.2.0 (required: https://github.com/tasmota/platform-espressif32/releases/download/2023.02.00/platform-espressif32.zip)
├── framework-arduinoespressif32 @ 2.0.7 (required: https://github.com/tasmota/arduino-esp32/releases/download/2.0.7/framework-arduinoespressif32.zip)
[... cut ...]
├── tool-esptoolpy @ 1.40001.220524 (required: https://github.com/tasmota/esptool/releases/download/v4.0.1/esptool-4.0.1.zip)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── tool-ninja @ 1.7.1 (required: platformio/tool-ninja @ ^1.7.0)
└── toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3)

```

diarus, 2023-06-24:
Thank you for your help. So, my Platfomio.ini :
`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
	gin66/FastAccelStepper@^0.30.4
	gyverlibs/EncButton@^2.0
upload_speed =  115200
monitor_speed = 115200
build_flags = -Werror -Wall -Wno-deprecated-declarations -Wno-error=incompatible-pointer-types
board_build.f_cpu = 240000000L`
**Result : the problem has disappeared**

diarus, 2023-06-24:
But, the problem remained in the Arduino IDE ver.2.1.0 :

`c:\Users\User\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp: In member function 'void StepperQueue::init_rmt(uint8_t, uint8_t)':
c:\Users\User\Documents\Arduino\libraries\FastAccelStepper\src\StepperISR_esp32_rmt.cpp:302:30: error: 'rmt_tx_memory_reset' was not declared in this scope
   rmt_tx_memory_reset(channel);
                              ^

exit status 1

Compilation error: exit status 1
[image]
 `

gin66, 2023-06-24:
The arduino library manager is completely broken since weeks and fixing the issue is slow. Moreover I do not use the Arduino IDE, because it cannot be automated in github actions AFAIK. So continuous testing every release is impossible and would require manual work, which I do not plan to spend/waste.

Consequently, it is hard to fix an error, which cannot be reproduced and perhaps is caused by poor and intransparent library handling. Even so the fix may be quite simple.

I can only recommend everyone to use platformio, eventually combined with Visual Studio Code.

diarus, 2023-06-24:
Ok, thank you.

closed 2023-06-24 by diarus