gin66/FastAccelStepper #186: Compatibility with external driver / period length?

neutebach, 2023-06-26:
Hi,

I'm trying to drive a not very complicated application (3D printing filament rewinder) that is basically just a linear guide with a NEMA17 and spindle the speed of which is calculated and updated multiple times a second, based on the speed that the winder is winding the filament. The speed in steps/s is calculated in an interrupt routine that is triggered by a sensor 6 times per rotation of the winder. 
I had the system working with another "library" for step generation that a colleague of mine wrote himself, so I cannot link to it, but I want to switch for various reasons. 

Long story short; the axis moves only very slowly, roughly and most notably only a fraction of the distance it's supposed to. The external driver is an igus D1 ([click for manual](https://igus.widen.net/content/lx2pnlx7jc/original/Operating-Manual-dryve-D1-EN-V3.0.1.pdf?u=gqh1pb&download=true)), in the manual it says that the max frequency it will accept is 25kHz **at a minimum of 40µs per step**. My first theory is that the period of each step generated by this lib is on the brink of being too short for the driver to register which means that a lot of steps are lost. Could this be the case?

thanks for having a quick look at this.
best regards, Niklas

gin66, 2023-06-26:
No issue with the period. The minimum length is not stated. There is only one issue mentioned, which may require to apply a direction pin delay (10us between dir change and step)

Questions:
- Which uC do you use ?
- Your uC can output 5V ?
- can you move your calculation out of the interrupt routine? Longer interrupt blockages can cause lots of issues.

neutebach, 2023-06-27:
thanks for your quick response.
I'm using a Controllino Maxi (based on Arduino Mega - ATmega2560)
It should output 24V.
I'll have a look at the interrupt routines and see if that makes a difference.

neutebach, 2023-06-28:
As I'm pretty sure the problem lies within my code, I'll close this issue. I don't have the time to dive into it unfortunately since it's a work project and it has worked before pretty ok, which means I can't justifiy putting more effort into it. Thank you though.

closed 2023-06-28 by neutebach