gin66/FastAccelStepper #187: arduino pro mini platformio project does not compile

alcho75, 2023-07-09:
hi,

i use VSC with platformio installed. i created a new project and use FastAccelStepper library. The project does not compile.

The errors:

Dependency Graph
|-- FastAccelStepper @ 0.30.4
Building in release mode
Compiling .pio\build\pro8MHzatmega328\src\main.cpp.o
Compiling .pio\build\pro8MHzatmega328\lib0f2\FastAccelStepper\FastAccelStepper.cpp.o
Compiling .pio\build\pro8MHzatmega328\lib0f2\FastAccelStepper\PoorManFloat.cpp.o
Compiling .pio\build\pro8MHzatmega328\lib0f2\FastAccelStepper\RampCalculator.cpp.o
In file included from .pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\FastAccelStepper.h:4:0,
                 from .pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\FastAccelStepper.cpp:1:
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h: In member function 'void ramp_config_s::update()':
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:23:25: error: 'pmfl_timer_freq' was not declared in this scope
 #define PMF_TICKS_PER_S pmfl_timer_freq
                         ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\PoorManFloat.h:32:31: note: in definition of macro 'pmfl_multiply'
 #define pmfl_multiply(x, y) ((x) + (y))
                               ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:173:29: note: in expansion of macro 'PMF_TICKS_PER_S'
       cubic = pmfl_multiply(PMF_TICKS_PER_S, cubic);
                             ^~~~~~~~~~~~~~~
In file included from .pio\libdeps\pro8MHzatmega328\FastAccelStepper\src/FastAccelStepper.h:4:0,
                 from src\main.cpp:8:
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src/RampCalculator.h: In member function 'void ramp_config_s::update()':
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src/RampCalculator.h:23:25: error: 'pmfl_timer_freq' was not declared in this scope
 #define PMF_TICKS_PER_S pmfl_timer_freq
                         ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src/PoorManFloat.h:32:31: note: in definition of macro 'pmfl_multiply'
 #define pmfl_multiply(x, y) ((x) + (y))
                               ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src/RampCalculator.h:173:29: note: in expansion of macro 'PMF_TICKS_PER_S'
       cubic = pmfl_multiply(PMF_TICKS_PER_S, cubic);
                             ^~~~~~~~~~~~~~~
In file included from .pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:6:0,
                 from .pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.cpp:1:
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h: In member function 'void ramp_config_s::update()':
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:23:25: error: 'pmfl_timer_freq' was not declared in this scope
 #define PMF_TICKS_PER_S pmfl_timer_freq
                         ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\PoorManFloat.h:32:31: note: in definition of macro 'pmfl_multiply'
 #define pmfl_multiply(x, y) ((x) + (y))
                               ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:173:29: note: in expansion of macro 'PMF_TICKS_PER_S'
       cubic = pmfl_multiply(PMF_TICKS_PER_S, cubic);
                             ^~~~~~~~~~~~~~~
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:23:25: note: suggested alternative: 'pmfl_ticks_h'
 #define PMF_TICKS_PER_S pmfl_timer_freq
                         ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\PoorManFloat.h:32:31: note: in definition of macro 'pmfl_multiply'
 #define pmfl_multiply(x, y) ((x) + (y))
                               ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:173:29: note: in expansion of macro 'PMF_TICKS_PER_S'
       cubic = pmfl_multiply(PMF_TICKS_PER_S, cubic);
                             ^~~~~~~~~~~~~~~
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h: In member function 'uint32_t ramp_config_s::calculate_ticks(uint32_t) const':        
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:24:39: error: 'pmfl_timer_freq_div_sqrt_of_2' was not declared in this scope
 #define PMF_TICKS_PER_S_DIV_SQRT_OF_2 pmfl_timer_freq_div_sqrt_of_2
                                       ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\PoorManFloat.h:33:29: note: in definition of macro 'pmfl_divide'
 #define pmfl_divide(x, y) ((x) - (y))
                             ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:203:23: note: in expansion of macro 'PMF_TICKS_PER_S_DIV_SQRT_OF_2'
           pmfl_divide(PMF_TICKS_PER_S_DIV_SQRT_OF_2, pmfl_sqrt_steps_mul_accel);
                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src/RampCalculator.h:23:25: note: suggested alternative: 'pmfl_ticks_h'
 #define PMF_TICKS_PER_S pmfl_timer_freq
[... 44 lines cut ...]
           pmfl_divide(PMF_ACCEL_FACTOR, parameters.pmfl_accel);
                       ^~~~~~~~~~~~~~~~
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h: In member function 'uint32_t ramp_config_s::calculate_ramp_steps(uint32_t) const':   
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:25:26: error: 'pmfl_timer_freq_square_div_2' was not declared in this scope
 #define PMF_ACCEL_FACTOR pmfl_timer_freq_square_div_2
                          ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\PoorManFloat.h:33:29: note: in definition of macro 'pmfl_divide'
 #define pmfl_divide(x, y) ((x) - (y))
                             ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:222:23: note: in expansion of macro 'PMF_ACCEL_FACTOR'
           pmfl_divide(PMF_ACCEL_FACTOR, parameters.pmfl_accel);
                       ^~~~~~~~~~~~~~~~
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h: In member function 'uint32_t ramp_config_s::calculate_ramp_steps(uint32_t) const':   
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:25:26: error: 'pmfl_timer_freq_square_div_2' was not declared in this scope
 #define PMF_ACCEL_FACTOR pmfl_timer_freq_square_div_2
                          ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\PoorManFloat.h:33:29: note: in definition of macro 'pmfl_divide'
 #define pmfl_divide(x, y) ((x) - (y))
                             ^
.pio\libdeps\pro8MHzatmega328\FastAccelStepper\src\RampCalculator.h:222:23: note: in expansion of macro 'PMF_ACCEL_FACTOR'
           pmfl_divide(PMF_ACCEL_FACTOR, parameters.pmfl_accel);
                       ^~~~~~~~~~~~~~~~
*** [.pio\build\pro8MHzatmega328\lib0f2\FastAccelStepper\RampCalculator.cpp.o] Error 1
*** [.pio\build\pro8MHzatmega328\src\main.cpp.o] Error 1
*** [.pio\build\pro8MHzatmega328\lib0f2\FastAccelStepper\FastAccelStepper.cpp.o] Error 1

gin66, 2023-07-15:
This is surprising. All compiles in github and on my local machine are using platformio. Just not from VSC, but from command line. It appears, as if the includes are not processed correctly.

gin66, 2023-11-08:
stale

closed 2023-11-08 by gin66