gin66/FastAccelStepper #188: Optional extra parameter for setExternalCallForPin()

Munger, 2023-07-09:
Could you please add an optional int32 parameter to setExternalCallForPin() which we can use to store a cookie value that is passed back in the callback? This would allow us to have per-instance callbacks, i.e. we can pass the 'this' pointer to setExternalCallForPin() and could retrieve it in the static callback.

gin66, 2023-07-15:
There is only one callback possible, which serves all external pins. This means for setting a cookie, another API call would be needed. Furthermore this change will break other app developers‘ code, which may not need this feature. So I am not convinced, that this change is a good choice for all users.

As normally steppers are statically allocated, a simple scheme could be to have a table with your int32 values and use the lower 7bits of the pin value as index. This means the pin values would be `PIN_EXTERNAL_FLAG+0…(n-1)` for n external steppers. As soon as the pin is marked as external, it is used only as unique reference. 

Even so it is not great to use this global table at all, but for an embedded system is IMHO acceptable.

Munger, 2023-07-16:
"There is only one callback possible..."

Where does that restriction come from?

"As normally steppers are statically allocated, a simple scheme could be to have a table with your int32 values and use the lower 7bits of the pin value as index. This means the pin values would be PIN_EXTERNAL_FLAG+0…(n-1) for n external steppers. As soon as the pin is marked as external, it is used only as unique reference."

That is certainly an option. I will look into it. Thanks for the suggestion.

gin66, 2023-07-16:
The callback belongs to the FastAccelStepperEngine class. For any stepper using a pin defined as external, the library will use that unique callback.

closed 2023-07-19 by gin66