gin66/FastAccelStepper #189: deprecated instruction

Elpablito, 2023-07-12:
I am reporting for yuor information a log of platformIO
StepperISR_esp32_rmt.cpp 2

 if (!q->_isRunning) {
      // second invocation to stop.
      rmt_tx_stop(q->channel);
      rmt_rx_stop(q->channel);
      rmt_memory_rw_rst(q->channel);
      q->_rmtStopped = true;
    }

[{
	"resource": "/c:/Users/gsrad/Documents/Lavori_con_Platform/ESP32 Nema23 Test/lib/FastAccelStepper/src/StepperISR_esp32_rmt.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'esp_err_t rmt_memory_rw_rst(rmt_channel_t)' is deprecated: use rmt_tx_memory_reset or rmt_rx_memory_reset instead [-Wdeprecated-declarations]",
	"startLineNumber": 283,
	"startColumn": 28,
	"endLineNumber": 283,
	"endColumn": 28
}]

digiexchris, 2023-12-13:
there are a number of deprecations in PCNT in esp-idf 5.0 as well (which arduino-esp is using under the hood), and are removed entirely in 5.1, such as PCNT_SIG_CH0_IN4_IDX. Eventually it'll crop up as a problem when arduino-esp moves to esp-idf 5.1.

gin66, 2024-03-09:
Reproduced in automatic builds. Need to be fixed

gin66, 2024-03-24:
The warning has been fixed in pre-0.3.12.

For esp32 just have commented the instruction out - seems to work. For esp32 derivates replaced with `rmt_tx_memory_reset()`.

closed 2024-03-24 by gin66