gin66/FastAccelStepper #190: the newest 0.30 has issue with pin 7, 8 on Mega2560. Pin 6 is OK

quanlotbangsat, 2023-07-14:
the newest 0.30 has issue with pin 7, 8 on Mega2560. Pin 6 is OK. Pls support me extend more stepper by using another timer

gin66, 2023-07-15:
Please check:
(https://github.com/gin66/FastAccelStepper#avr-atmega2560)

Could you please describe the issues on pin 7,8 ? As I do not own this device, I perform tests only on simulator and check for running even three steppers with atmega2560 and there is no test failure popping up.

quanlotbangsat, 2023-07-19:
can you real test with mega board? I'm using it and have problem when using real hardware. And can you check when the speed change the distance is change? I don't know why. When i using the screw with speed 25000 sps the distance is diffirent with 1000sps.

closed 2023-07-27 by quanlotbangsat