gin66/FastAccelStepper #191: about the arduino 328P setSpeedInHz max only 1000hz using version 0.30.0

LP-ming, 2023-07-15:
hello ,First of all, thank you for the library。
     I use your library to control stepper motors，the Control board i am using the arduino 328P.steppin 10,dirpin 11,enablepin 12.The pulse frequency I need is about 20000Hz~70000Hz.
     When I use version 0.28.4 of the library, the 328P is able to pulse at the frequency I need,But when I upgraded to 0.30.0, I found that the 328P could not pulse at the frequency I wanted。The maximum frequency of the pulse at 328P can only be up to 1000Hz。
      Do I have any settings, or the parameters are not configured correctly？Looking forward to your answer, thank you。I can provide my code if needed

gin66, 2023-07-15:
The new version does not allow 70000steps/s, but limits to 50000 steps/s using only one stepper.
„ allows up to 50000 generated steps per second for single stepper operation, 37000 for dual stepper and 20000 for three steppers“.

In any case, would be good to have a look at the code for clarifying, why this slow speed comes out.

LP-ming, 2023-07-18:
Thanks for your reply。Unfortunately, I didn't have to check for any errors from the code。but i found that after  writing issue173 to 328P, PIN10 can PULSE 50KHZ .So, I'm going to rewrite my program based on issue173。3Q

gin66, 2023-07-18:
btw: which version of FastAccelStepper do you use ?

This could be related:
  0.30.4: Fix for issue #178: speed does not decelerate but jumps to lower value

gin66, 2023-07-18:
Yes, the Arduino Library Manager is completely broken and gets not fixed since months.
Good time to switch to something more professional like platformio.

closed 2023-11-08 by gin66