gin66/FastAccelStepper #192: the decelerating at RunForward blocks briefly

HerrFlodo, 2023-07-17:
Hi,

I have a problem with the deceleration in the RunForward command. I change the speed with a potentiometer in the loop as follows:

void loop() {

  potValue = analogRead(potpin);
  targetSpeed = map(potValue, 0, 1023, 1, 40000) ;// Targetspeed pro Intervall
    
    stepper->setAcceleration(30000);
    stepper->setSpeedInHz(targetSpeed);  // the parameter is Hz !!!
    stepper->runForward();
}

The acceleration works wonderfully, but with the deceleration I have errors. There seems to be no acceleration function behind the deceleration, and when I turn the potentiometer slowly the stepper sometimes brakes briefly to a standstill and then accelerates again.

Can anyone help me with this problem?

Greetings

Flo

HerrFlodo, 2023-07-18:
Hi Gin66,

Thanks a lot for the very fast response. I will try that in the afternoon. What I am also confused of is the missing deceleration function. The stepper speed follows direkt the potentiometer position. It should not behave like this, right? If I speed up, it works like expected.

HerrFlodo, 2023-07-18:
Hi,

so i tested the above code today.

int average = 0;

void loop() {
   potValue = analogRead(potpin);
   average += (potValue - average)/4;
   targetSpeed = map(average, 0, 1023, 1, 40000) ;// Targetspeed pro Intervall

   stepper->setAcceleration(30000);
   stepper->setSpeedInHz(targetSpeed);  // the parameter is Hz !!!
   stepper->runForward();
   delay(1);
}

 Unfortunately it does not bring any improvement. I am using version 0.30. It is conceivable that I have the same error as described in #178, but I could not reproduce it with the code in issue #178.

gin66, 2023-07-18:
Thanks for the update.

So there are two possibilities:
1. You stick with your solution, which works for you.
2. Manually update to 0.30.4 and hope, that fix works in your situation, too.

While you are at it, perhaps you leave a comment [here](https://github.com/arduino/library-registry/issues/2829)