gin66/FastAccelStepper #193: Mega 2560 - 0.28.4 - The distance diffirent after change speed

quanlotbangsat, 2023-07-19:
Hi gin66, thank you so much for your lib.

But i have issue when using mega 2560, first it maximum 1000hz if i using your 30.4 library, So i must using 28.4

On 28.4, I don't know why when i change speed the distance will diffirent. When the speed is slow the distance go longer.  The code using Zero like this and moveto:
bool zeroservo(FastAccelStepper *servo, ServoMM *servodata, byte zerosensor,
		float speed) {
	bool done;
	if (!servo->isRunning()) {
		if (digitalRead(zerosensor)) {
			servo->setSpeedInHz(speed);
			servo->move(-1000000);
			onzero = true;
		} else
			goto d;

		while (digitalRead(zerosensor)) {

			if (!servo->isRunning())
				goto a;
			if (stopnow == 1)
				goto b;
			loop();

		}
		d:
		//servo->stopMove();
		servo->forceStopAndNewPosition(0);
		while (servo->isRunning())
			;
		//Serial.println(servodata->zeropoint_offset);
		servo->move(servodata->zeropoint_offset);

		while (servo->isRunning()) {

			loop();
		}
		servo->setPositionAfterCommandsCompleted(0);
		servodata->homed = 1;
	}
	onzero = false;
	return 1;
	b: stopnow = 0;
	servodata->homed = 1;
	onzero = false;
	a:

	return 0;
}

Pls support me

gin66, 2023-07-19:
For better understanding: Where is `zeroservo()` called ? Inside `loop()` or in `setup()` ?

quanlotbangsat, 2023-07-20:
I call it throught serial.

gin66, 2023-07-20:
not clear. Could you please provide more of your code ?

There are calls to `loop()` within this function. This smells like a recursive use of `loop()`, which may cause all sorts offside effects.

closed 2023-07-27 by quanlotbangsat