gin66/FastAccelStepper #195: ESP32 - engine.init() Core selection not working!?

j0k3r187, 2023-08-13:
Hello,
first of all, thanks for this Library. 
Im trying to run this library with a Async Webserver but after starting the Server the Steppermotor is stuttering a lot.
Like its not getting precise Step signals. Im using an ESP32-WROOM 32, but testet on different ESP32 DevkitC V4
Modules with the same result.

I thought if i use Arduino Core: 0, Event Core: 0 and engine.init(1) then my Stepper is running on Core 1 and the Webserver on Core 0.
But no matter what combination of  Arduino Core, Event Core and engine.init() the Stepper keeps stuttering.

Here is an example to repoduce this:

Testrun 1 is running without any problem.
After starting a simple Server Testrun 2 is Stuttering.

<details>
<summary>Testprogramm</summary>

#include "Arduino.h"
#include "WiFi.h"
#include "AsyncTCP.h"
#include "ESPAsyncWebServer.h"
#include "FastAccelStepper.h"

///// Netzwerk Parameter
const char *AP_ssid = "ESP32_MotorControl";
const char *AP_password = "ADMIN123";

const char *http_username = "w";
const char *http_password = "w";
  
IPAddress local_IP(192, 168, 0, 187);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 0, 0);
IPAddress AP_subnet(255, 255, 255, 0);
IPAddress primaryDNS(8, 8, 8, 8);   // optional
IPAddress secondaryDNS(8, 8, 4, 4); // optional

AsyncWebServer server(80);

///// Pin´s

int EN_PIN = 4;
int STEP_PIN = 16;
int DIR_PIN = 5;

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  Serial.begin(115200);
  

  engine.init();
  
  stepper = engine.stepperConnectToPin(STEP_PIN);
  if (stepper) {
    stepper->setDirectionPin(DIR_PIN);
    stepper->setEnablePin(EN_PIN, false); // (Enable Pin, LOW Active)
    stepper->setAutoEnable(false);
    stepper->enableOutputs();

    stepper->setSpeedInUs(300);
    stepper->setAcceleration(10000);
  }

  Serial.println("Start Testrun 1 . . .");
  stepper->runForward();                 
  delay(2000);
  stepper->forceStop();
  
  ///// Server Setup
  if (!WiFi.softAPConfig(local_IP, gateway, AP_subnet))
    {
      Serial.println("WiFi AP Config failed!");
    }
    // Start in AP Mode
    Serial.println("Starting in AP Mode!");
    if (!WiFi.softAP(AP_ssid, AP_password))
    {
      Serial.println("WiFi AP Mode failed!");
      return;
    }
    else
    {
      Serial.println(WiFi.softAPIP());
    }

  // Start server
  server.begin();

  Serial.println("Start Testrun 2 . . .");
  stepper->runForward();
  delay(2000);
  stepper->forceStop();
}

void loop(){
  delay(1);
}
</details>

ps. The delay(1); is needed if Arduino Core runs on Core 0, otherwise ESP keeps Rebooting

gin66, 2023-08-13:
seems related issues:
[The WiFi task suspends other tasks on both cores of ESP32](https://github.com/espressif/arduino-esp32/issues/4176)
[ESP32 hardware ISR sometimes not triggered when wifi is transmitting](https://stackoverflow.com/questions/71921533/esp32-hardware-isr-sometimes-not-triggered-when-wifi-is-transmitting)

The preconditions for FastAccelStepper to run smoothly are IMHO:
* no suspension/blocking of the stepper task
* no other interrupts blocking the cpu for longer than the fastest time step
* no writes to the flash e.g. OTA/writing, because the stepper task code is in flash. writes <4ms may be ok.

No idea, how to make the WiFi driver more cooperative.

Another user has used bluetooth successfully with FastAccelStepper, if the Wifi issue cannot be resolved.

gin66, 2023-08-13:
FreeRTOS is a cooperative OS, so a `loop()` without task switch will block the core. If arduino core is running on core 0, then the watchdog will not serviced anymore and trigger a watchdog reset. AFAIK on core 1, there is no watchdog.

Instead of `delay()` you can use `taskYIELD()`.

gin66, 2023-11-08:
stale

closed 2023-11-08 by gin66