gin66/FastAccelStepper #196: migrating from AccelStepper

guestisp, 2023-08-16:
Hi all.
i'm planning the migration from accelstepper trying to reach higher speed on a ESP32.

I'm not using accelerations or anything else except basic movement commnad, is this a sort of drop-in replacement for accelstepper or should I adjust the code ?

gin66, 2023-08-19:
Not exactly drop in, as the API has differences. Nevertheless adaptation should be straightforward

guestisp, 2023-08-19:
uhm non exactly straightforward as i have to recalculate all speeds, as currently i'm using speeds in steps*sec but this library wants a different value.

also, i didnt find anything to check the current position over the movement, in example step 78 over 150

gin66, 2023-11-08:
stale

closed 2023-11-08 by gin66