gin66/FastAccelStepper #198: Info: Using FastAccelStepper for PVA execution

aaryanmurgunde, 2023-08-29:
Hi,

Scenario : I have a 6 DoF arm that I would like to control with the library, I have high level programs with ROS and MoveIt that generates a trajectory for execution with ROS Control and HW Interfaces. I want have an ESP32 (preferable one for 6 motors) execute the points of the trajectory. On the ESP end I receive points in the from of -> (position, velocity, acceleration). The acceleration and the velocity profiles for accelerating and decelerating are generated in the trajectory itself. 
If I choose to use the inbuilt acceleration and decelerating feature what happens is a jitter is faced in between the points of the trajectory as the lib chooses to get the motor to a stop.

What my ideal scenario looks like: I get the point from ROS (including the acceleration), set it using the library and have some kind of notification to queue the next point on the motor. The points are also based on time i.e. The points are generated for synchronised motion of all the motors.

I'm pretty new to this lib and have not have a successful base implementation as well. Also if you know any better ways to implement the above scenario, please suggest.

Any help is appreciated.

Thank You.

gin66, 2023-08-29:
With the built-in ramp generator, this will be quite difficult to achieve. If possible at all. A workable solution would be to use raw stepper commands and send them to the respective steppers‘ command queue. This means, that there is no acceleration/deceleration support from FastAccelStepper. As you have already (position, velocity, acceleration), can you change your high level system to issue (position, speed) for all  steppers every 1ms and calculate acceleration/deceleration on your own ? A stepper command is basically: direction, number of ticks and rate of ticks.

aaryanmurgunde, 2023-08-30:
@gin66 Yes definitely that is one approach that I'm doing right now. This is was an upgrade I was planning to implement

gin66, 2023-11-08:
stale

closed 2023-11-08 by gin66