gin66/FastAccelStepper #199: External autoEnable also controls pin 0

Carl-UB, 2023-09-01:
When using an external enable pin, and with autoEnable on, running the stepper motors also controls pin 0.

[Failing test & workaround with test here.](https://github.com/Carl-UB/FAS_ExternalAutoEnableBug)

Seems likely to be due to the `_enablePinHighActive` being 0 by default and not being changed when setting enable pins. Could it be instead initialised to `PIN_UNDEFINED`? It should still then be able to be set to a new value by the user but would avoid this bug.

Workaround is to manually call `stepper->setEnablePin(PIN_UNDEFINED, false);` to set `_enablePinHighActive` manually.

I presume that `_enablePinLowActive` could have the same sort of problem, I haven't tested that.

Carl-UB, 2023-09-01:
Issue actually also occurs without external pins being used. Have updated the example repo and added tests using internal pins as well.

gin66, 2023-09-01:
Yes. Apparently the initialization of the internal variables `_enablePinHighActive` and `_enablePinLowActive` are missing.

gin66, 2023-09-01:
Could you please check, if this patch works as expected ?

Carl-UB, 2023-09-12:
Yep, looks like that has worked. Thanks!

closed 2023-09-12 by Carl-UB