gin66/FastAccelStepper #202: Stepper motor unable to exceed 1 rpm without vibrating

skemp117, 2023-10-07:
Hey, this is such a cool library. Thanks for making it.

I'm getting heavy vibrations and stalling as I increase the revolutions per second `stepper->setSpeedInHz(rps*3200);` It was struggling above 2ish, gets extremely rough around 5, and stops entirely around 7. I suspect that the shaft speed isn't actually as high as it is being set to, it feels like it is only doing 2 ish rps.

I'm using BIGTREETECH TMC2209 drivers at 16x microstepping on a Azteeg X3 Pro board. The step pin is 46, and I have the `build_flags = -Werror -Wall -DFAS_TIMER_MODULE=3` set. I'll attach the code below. I appreciate any help. I've tried changing both the motor and stepper drivers many times. I've double checked the VRef, and it is set correctly. The drivers are old, maybe they just suck.

`#include "FastAccelStepper.h"

#define ENABLE_PIN 62
#define DIR_PIN   48
#define STEP_PIN  46

#define MICROSTEPS 16
#define STEPS_PER_REV 200 //1.8 degrees per step, 3200 steps per rev at 16 microsteps

#define MICROSTEPS_PER_REV MICROSTEPS*STEPS_PER_REV

int step_cnt = 0;

FastAccelStepperEngine engine = FastAccelStepperEngine();
FastAccelStepper *stepper = NULL;

void setup() {
  engine.init();
  stepper = engine.stepperConnectToPin(STEP_PIN);
  if (stepper) {
    stepper->setEnablePin(ENABLE_PIN);
    stepper->setAutoEnable(true);
    stepper->setDirectionPin(DIR_PIN);

    stepper->setSpeedInHz(1*3200);       // steps/s
    stepper->setAcceleration(200);    // steps/s²
    stepper->runForward();
  }
}

void loop() {
    // stepper->move(3200*10);
}`

gin66, 2023-10-08:
There could be several reasons:
* FastAccelStepper has been only tested with timer 5 on a simulator. Any chance to try with standard 4 and pin 6,7,8 ?
* What is the power supply voltage/current rating ?
* Which cables/ cable length between power supply and Azteeg ?
* Could the mechanical load be the root cause ?
* …

For investigation:
* Do you have an oscilloscope available to check the generated steps ?
* For lower rpm you could cross check with AccelStepper or you make a handwritten loop to manually toggle the step pin. If so, is the issue gone ?
* Which library version do you use ?

skemp117, 2023-10-10:
Thanks for the detailed reply!

- The Azteeg X3 Pro board that I am using has a very unfortunate pinout. Only one pin with a timer is available (46) to the motor driver slots. I could try getting access to another pin to hardwire to the step pin on the motor drivers, but I'd probably just buy another board first.
- 12V at 50 Amps (it is way overkill lol, got it from a factory)
- Cables are reasonable short, maybe like 1 foot, high gauge wire.
- No load on the stepper at all.

- I could get access to one, but I'd rather just buy a new board
- Low RPMs work fine with Accel stepper
- 30.5
- https://github.com/gin66/FastAccelStepper/assets/44552792/56874ad1-dc2b-4911-b0fc-3e14a7d98863

Do you recommend any small footprint boards you'd recommend. Maybe like the bigtreetech skr mini e3 v2.0?

gin66, 2023-10-10:
Thanks for all the answers and sharing the video. This is definitely not the expected behavior and actually the NEMA-17 is running at very low speed. Just have checked with simulator (simavr) up to 10kSteps/s and all is as expected. As per issue #50, one user has reported timer module 5 to work. 

Assume the NEMA-17 is not broken, then I am currently running out of ideas, what the problem could be. Sorry for this.

skemp117, 2023-10-10:
Do you recommend any small footprint boards you'd recommend. Maybe like the bigtreetech skr mini e3 v2.0?

gin66, 2023-10-10:
I cannot recommend any of the boards out there. For FastAccelStepper development I have made an own pcb, where I can mount an arduino nano and an esp32dev at the same time and can control in total eight stepper motors using A4988 drivers. Perhaps one of the users can provide you with recommendations.

gin66, 2023-11-21:
stale

closed 2023-11-21 by gin66