gin66/FastAccelStepper #205: Using library with no direction pins connected

jackthese, 2023-10-15:
I have an application using Trinamic 2209 drivers, using the option to set direction by UART, not connected direction pins. Am I correct in assuming that there is no way that the library can keep track of the stepper position? My workaround is to use only move commands (and not moveTo), and do my own arithmetic to keep track of where the motor is, taking into account which moves are in a CCW direction, and which are CW. Just curious to know if there is a better way. ...

gin66, 2023-10-15:
you can define a callback for a pin with `setExternalCallForPin()`. Works for enable and direction. This way, FastAccelStepper can keep track of the position.

jackthese, 2023-10-15:
And I presume it doesn't have to be a real pin, and nothing gets connected? I'm out of pins for this project, and used UART to save 4 pins used elsewhere. Thanks for the tip so quickly!

jackthese, 2023-10-15:
OK, as I noodled through the example, it seems that I would have to use four real unique pins, and I simply don't have enough available for this application. My workaround meets my needs, but my implementation is far short of the power in the library (my moves must always complete for my calculation to be valid, for instance).

gin66, 2023-10-15:
It does not have to be a real pin.  The pin number is identified with the bit PIN_EXTERNAL_FLAG (=128) being set. This means, you can use e.g. 128, 129, 130, and 131  as pin numbers for the direction pins. FastAccelStepper does not look at the value at all, if PIN_EXTERNAL_FLAG is set (except for shared enable pins). So your application will be called with those pin numbers 128…131. In the callback you can derive your UART pin by `pin & 3` and just perform the set high/low action.  In case your callback is not fast, then better to  just set a flag and update the UART in the remaining application code.

jackthese, 2023-10-15:
Wow, I did exactly as you suggested, and it works perfectly. Had to replace the sample set high/low action with a UART call (	BigTree[pin].shaft(myDir); ), and the motor moved correctly and reported its position correctly. Thanks for helping out today. Above and beyond expectations...Jack

jackthese, 2023-10-15:
I consider this closed, thanks again

closed 2023-10-15 by jackthese