gin66/FastAccelStepper #206: Cant get higher speed than 8000 (Arduino due)

TheOnlyBeardedBeast, 2023-10-21:
Hi, I am running the newest version on Arduino DUE, I am running in 1/8 microstepping with TMC2209. I cant get a speed over 8000, no matter if I setup slow acceleration, or a fast one I get the same result. 8000 works nice but when I tested 9k,10k,12k hz it just vibrating, the motor accelerates then it got to a high speed, and it just start moving back and forth really fast but still moving a little bit in one direction. like it is losing steps. The issue is definitely not the motor, as on the DUE I am still able to run AccelStepper with these speeds, tested up to 20k with 200k acceleration. But it would be nice to migrate to FastAccelStepper as I need to run 4 motors in these speeds.

I setup the tested motor at pins
STEP pin 6
DIR pin 7
an another motor was setup at pins, but was not set any instruction, after I saw what the motors did in my project when I switched AcceelStepper to FastAccelStepper
STEP pin 8
DIR pin 9

Thanks

clazarowitz, 2023-11-06:
Hmmm, I’ve definitely run faster than 8khz.  I suspect you’ve got more interrupts running on the processor.  That’s not to say that this code is behaving.  I documented in another post that I want to rewrite to only use the pwm interrupts.  I suspect that will fix a lot of issues.  There are some other optimizations to be had as well.  So this is not a works for me…not at all.  In fact, I’d like to know if you run just the steppers, does it work?  If it’s just barely not working, I’d bet changing the PIO interrupt to falling edge and killing the usleep would make something of a difference.  How much?  Not sure.  But I can probably make that change sooner than the whole rewrite. 

That said, I do want to get to the rewrite.  However, the project I ported for initially is still dragging on.  (Not the software part). I should put up a project log somewhere…I’m fairly certain though, I’m done with the mechanicals.  So I should be back to the software soon.  In fact, if the latest change doesn’t fix the “backlash” I’m seeing, I’ll have to assume it’s real.  But I’d found a sensor was able to move…so, it’s unlikely.  

so, my intended schedule is to get this thing tested this week, next week be some more integration testing, and hopefully the week after being back to some serious software work!

TheOnlyBeardedBeast, 2023-11-06:
@clazarowitz no other interupts run on the arduino, it was just a test code for fast accelstepper, but I found out accelstepper was lying about the achieved speed, but it was still more than 8k, 12k was meassured. I moved to arduino giga and also wrote my timer based stepper micro library. But would like to use the due in the future. Sorry to bother. I tried to look into the source of this lib but to be honest I dont understand the timer stuff (I am new in this lower level programming), also chceked out DueTimer so maybe it could help, but all I got was was stable slow speed. On Giga I combined portenta timer with some port manipulation to control multiple steppers at the same time with Bresenhams algorhitm, and that did the magic for me, but I again hit the 12k steps/sec limit. I tried multiple ramping algos but 12k was the max I got. So for now I will stick to the giga and ordered some new motors with defined torque curves, but I can try the new motors on the Due and fastaccellstepper as well if this type of test would help you.

gin66, 2025-02-20:
stale